ArduPilot/ardupilot #27929: #26740 causes old params to appear on Wiki param pages

Hwurzburg, 2024-08-26:
This has caused a user to raise wiki issue stating the wiki pages are not using the correct params, since both the old and new names aare shown in the emitted parameter pages...we have not done this for previous param renames...either we show the old names somehow differentiated in the wiki param page or we should revert this....I dont see how NOT doing this was a "breaking" change to older firmware as PeterB told me...

rmackay9, 2024-08-26:
I've slightly modified the description of this issue to include the link to the PR in question https://github.com/ArduPilot/ardupilot/pull/26740

IamPete1, 2024-08-26:
We have tried to do this where we remember. The problem is that Mission planner uses the master param descriptions, so we were getting issues that the param descriptions where not there for GPS. For example (https://github.com/ArduPilot/ardupilot/issues/26681)

I think having too many params is better than having too few, until we fix the GCS side I'm not sure how we can do it better.

Hwurzburg, 2024-08-26:
can we add a note in the param metadata to warn that the param is renamed as of version x.x?

peterbarker, 2024-08-27:
Was fixed

closed 2024-08-27 by peterbarker