ArduPilot/ardupilot #27975: SITL crashes with new version of RealFlight Evolution

VermennikovMV, 2024-09-01:
## Bug report

**Issue details**

Previously, the SITL connection to RealFlight Evolution worked correctly, but (as I understand it) RealFlight was updated on August 28. After this update, when you connect SITL, the game stops working after a couple of seconds (freezes). When you disable SITL, it starts working again. 
The problem has been tested on two computers. The SITL connection was working properly exactly until Steam started updating RealFlight, then the game crashed. After a full update, SITL caused the game to crash.

**Version**

I think it was the latest version of SITL (since it seems to have auto-updates) and the latest version of RealFlight Evolution.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying wing

**Hardware type**
None

**Logs**
There was no way to get

IamPete1, 2024-09-03:
I have just tested here with the [SkyCat](https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/Released_Models/QuadPlanes/Tailsitters/SkyCat_TVBS) and the latest updates, it works fine for me. 

I'm not sure what to suggest, can you upload your model and params maybe the problem is related to the exact model. Or you could try something from our models [repo](https://github.com/ArduPilot/SITL_Models/tree/master/RealFlight/Released_Models).

Georacer, 2024-09-10:
I think you need to ensure SITL is **NOT** running when you launch RealFlight.