ArduPilot/ardupilot #27984: What's the path forward towards having 64bit builds for linux?

Williangalvani, 2024-09-02:
Navigator (and I'm pretty sure all linux boards) still runs on 32bit OSs.
We are putting effort into migrating [BlueOS](https://blueos.cloud/) to Latest Raspberry Pi OS (Bookworm) and 64bits.

64bit builds already work when built manually (at least musl), but how do we deploy them?

Should we add a new board? would that mean each linux board with 64bit support gets duplicated?
Or maybe we could just add two binaries to the same folder? both `ardusub` and `ardusub_64` for example?

Williangalvani, 2024-09-03:
we discussed in the call we can try to use static builds, and if that doesn't work, we should be able to either install the required libs for running the 64bits on raspbian, or create a new navigator board target for the builds.

peterbarker, 2024-11-24:
Given we now have navigator64 - I'm going to close this one now :-)

closed 2024-11-24 by peterbarker