ArduPilot/ardupilot #28068: Publish Gimbaled LRF distance through mavlink camera protocol

dirksavage88, 2024-09-10:
## Feature Request
Publish Gimbaled LRF distance through mavlink, instead of just the backend mount message

**Is your feature request related to a problem? Please describe.**
Currently the gimbal LRF distances for the viewpro and SIYI are published via an ardupilot specific mount message, see https://github.com/ArduPilot/ardupilot/blob/c40422cde32cd3b78630fcf277fa9dce4486e1e3/libraries/AP_Mount/AP_Mount_Backend.cpp#L477

However this implementation just passively publishes the distance values, whereas these range values can be consumed by some (LightwareLidar is one such gimbal LRF) mavlink-enabled gimbal/camera payloads to focus on a POI and would be a valid use case for these LRF values given the mavlink camera protocol v2.

**Describe the solution you'd like**
Instead of only publishing a custom mount message specific to ardupilot, publish via [MAV_CMD_SET_CAMERA_FOCUS](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_FOCUS) with focus type set to range and the value in meters from the POI

**Describe alternatives you've considered**
modifying the current implementation to publish gimbal LRF values to[ MAV_CMD_TRACKING_GEO_STATUS](https://mavlink.io/en/messages/common.html#CAMERA_TRACKING_GEO_STATUS), however the camera protocol/manager would have to be actively tracking to send distance values for focusing?

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Willing to contribute to push this forward, but I am new to ardupilot contributions.

rmackay9, 2024-09-10:
Hi @dirksavage88,

Thanks for the enhancement request.  Technically it's easy to add the rangefinder distance to a mavlink message, the issue is that I don't think we really have a good message at the moment.  MAV_CMD_SET_CAMERA_FOCUS and MAV_CMD_TRACKING_GEO_STATUS are both commands so those probably aren't appropriate.

It's probably best if we raise an issue on the [MAVLink issues list](https://github.com/mavlink/mavlink/issues) and get input from @hamishwillee and others.

hamishwillee, 2024-09-10:
I don't understand where the information is coming from or going to, but rangefinders are expected to publish information using https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR . Posting an issue as Randy suggests is a good idea (if this doesn't do what you want) so that I can share more broadly among mavlink users.

geofrancis, 2024-10-24:
Would it be better to add a new rangefinder orientation for gimbal mounted? that way it can be sent as a standard distance sensor. 

31 for gimbal 1
32 for gimba 2?

dirksavage88, 2024-10-24:
It would be one of these? https://mavlink.io/en/messages/common.html#MAV_SENSOR_ORIENTATION

Then obstacle distance sensor or terrain height/range aid measurements could ignore distance sensors with that orientation?