ArduPilot/ardupilot #28101: Can't set 100Hz for MAVLINK_MSG_ID_HIGHRES_IMU on Rover

lida2003, 2024-09-14:
**Setup:**

- latest 4.6-dev
- 921600 (UAVT, MAVLinkv2)

same code from setting MAV_CMD_SET_MESSAGE_INTERVAL,MAVLINK_MSG_ID_HIGHRES_IMU to 100Hz, which is 10000 us.

**Result:**
- F435 Copter 100Hz OK
-  H743 Copter 100Hz/200Hz OK
- H743 Rover NG about 50Hz

PS: Previous discussion here[Can't set 100Hz for MAVLINK_MSG_ID_HIGHRES_IMU?
#4](https://github.com/chobitsfan/mavlink-udp-proxy/issues/4).

stephendade, 2024-09-17:
Just a guess, but the loop rate in Rover ``SCHED_LOOP_RATE`` is 50 by default. Trying increasing this to 100 or 200.

lida2003, 2024-09-17:
I successfully set 10Hz(actually 9Hz), 25Hz(actually 24Hz), 50Hz(actually 49~50Hz), but I can't set 40Hz, 100Hz, 200Hz.

But it works for copter code 4.6-dev(same git commit code). It might be copter handles this command different from rover. I'm NOT digging into the code right now. 

Just post this question here, maybe someone can take care of it when have time.

stephendade, 2024-09-18:
Ok, I've done some digging and created a patch in #28144.

I've confirmed by initial guess that Ardupilot is indeed capping the message rate at ``SCHED_LOOP_RATE``. Try setting this parameter to 250Hz and try your message rates again.

lida2003, 2024-09-18:
I'll do the test tomorrow and get back to you soon.

stephendade, 2024-09-20:
> Is there any api for changing "SCHEDULER_LOOP_RATE"?

That's just an Ardupilot parameter that you can set via your GCS.

lida2003, 2024-09-20:
Thanks.
[image]

closed 2024-09-21 by lida2003