ArduPilot/ardupilot #28120: Sub: pilot override in POSHOLD mode results in "bounce back"

clydemcqueen, 2024-09-16:
## Bug report

**Issue details**

Pilot override in POSHOLD mode results in a ~1m "bounce back".

I saw this on real hardware (BlueROV2 heavy + DVL) running a DEV build. I moved the ROV away from the dock while in POSHOLD mode, when I let go of the stick the ROV seemed to jump back towards the dock.

I can reproduce this in SITL on master and Sub-4.5. I was not able to reproduce on Sub-4.1.

To reproduce:
* run sim_vehicle.py with -w, etc.
* run QGC w/ joystick
* wait for _EKF3 IMU0 is using GPS_
* arm, poshold
* use the joystick to move forward aggressively a few meters
* let go of the stick, the sub will move back a meter or so

**Version**

master, Sub-4.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ x ] Submarine

**Airframe type**

All

**Hardware type**

All

**Logs**

I can provide the BlueROV2 logs, but they are a bit big (replay on, long-ish guided mode test). LMK.

Williangalvani, 2024-09-18:
I could replicate this on Sitl, but not on a real vehicle. 
On SITL the target_position keeps changing when we let go of the sticks:
[image]

This doesn't seem to happen on my rov:

[image]

clydemcqueen, 2024-09-18:
This is from a dive with a BlueROV2 heavy w/ DVL on 2024-sep-11: 
[image]

I'll try to dig into the code tomorrow.

clydemcqueen, 2024-09-19:
Huge thanks to @Williangalvani for tracking this down!

It looks like a few factors contributed:
* Sub poshold mode calls `position_control->set_max_speed_accel_xy` with WPNAV_SPEED. In our case, we set WPNAV_SPEED very low for running survey transects.
* Sub poshold does not limit the pilot xy speed. In our case, the pilot was easily able to exceed WPNAV_SPEED.
* When the pilot let go of the stick, the PSC velocity controller took over and quickly wound up until acceleration hit the max, and the sub lurched backward.

Proposed fix:
* Add a PILOT_SPEED parameter for xy pilot control, and use it in poshold.

This is important for our use case so we are motivated to fix it. But I suspect that most users have never experienced this, and probably won't.

PR #28205 (2025-02-03): Sub: limit poshold xy velocity to PILOT_SPEED to avoid bounceback
This PR limits horizontal velocity in poshold mode, even when the pilot is using the forward and lateral sticks to reposition the sub. This eliminates the possibility of a PSC-caused bounceback. This is verified by an autotest.

The PR also adds a PILOT_SPEED parameter, so there are now 3 WPNAV_SPEED* parameters and 3 PILOT_SPEED* parameters. Previously, poshold was using WPNAV_SPEED for xy and PILOT_SPEED_UP/DN for z, this PR changes it to use PILOT_SPEED* for both xy and z.

(Circle mode also uses WPNAV_SPEED for xy and PILOT_ [... cut]
closed 2025-02-03 by Williangalvani