ArduPilot/ardupilot #28126: BATTERY_STATUS message stopped streaming

twistedwings, 2024-09-17:
## Bug report

The BATTERY_STATUS message stopped and continued streaming during flight.
All other messages seem to stream continuously.

Gaps in UAV Log Viewer
[image]

**Version**
4.5.6

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Matek F405-CTR

**Logs**
[Logs.zip](https://github.com/user-attachments/files/17022869/Logs.zip)

peterbarker, 2025-05-09:
@twistedwings we think this might be because of a telemetry link which can't handle all of the data that wants to be sent.

Certainly reducing the telemetry rates seemed to fix it locally.

I'm not 100% convinced this is the entire story, but please do try reducing your telemetry rates and seeing what happens.

twistedwings, 2025-05-12:
the Matek F405-CTR has been replaced by a MicoAir743.
I cannot reproduce the bug anymore.
Maybe it was just a lack of power.

rmackay9, 2025-05-12:
@twistedwings, great, thanks for the feedback.  I'll go ahead and close this then, thanks!

closed 2025-05-12 by rmackay9