ArduPilot/ardupilot #28164: Copter: RELAY output gives short HIGH during boot

fingadar, 2024-09-20:
## Bug report

**Issue details**
There is a change in RELAY output from 4.4 to 4.5.6. During boot process, the FC outputs a short high level at RELAY(x) pin, even if RELAY(x)_DEFAULT is zero. When booting is complete, the RELAY(x) pin is low again.
The effect is reproduceable, V4.4 → output stays low, V4.5.6 output has a short high pulse. 

**Version**
4.5.6

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek H743 slim V3.

rmackay9, 2024-09-20:
This might be one for @IamPete1 who I think did some refactoring of the relay code

IamPete1, 2024-09-21:
@fingadar Can you post your parameters please, which pin are you using?

rmackay9, 2024-09-23:
Hi @IamPete1,

I'm think I've bumped into this issue as well with my Xacti camera which has Relay1 connected to the camera's on/off pin ([see wiki here](https://ardupilot.org/copter/docs/common-xacti-gimbal.html#connecting-to-the-autopilot)).  Pulling the pin high causes the camera to reboot and this happens each time I reboot the autopilot using a mavlink reboot command.

I can manually get around the problem by using MP's Servo/Relay tab and setting the toggle high (to reboot the camera) and then low again but it would be nice if it didn't happen because I need things to go smoothly during my competition in a few weeks.

I went back to check 4.4.4 but my Xacti camera driver was not available then so for me this is not a new problem.

fingadar, 2024-09-23:
@IamPete1 
Yes, of course. Do you need the whole parameter file?
The important entries are:

`RELAY_DEFAULT,0` in 4.4

`RELAY2_DEFAULT,0` in 4.5.6

`RELAY_PIN2,60`

The output is S11 (GPIO60) on a MATEK H743 slim. The high toggle always happens during boot. It happens always at power-on, but also at a reboot using mavlink rebooting command.

In our case, the relay pin works in conjunction with a switching logic for a dual fc configuration, and so, on reboot or power-on, the logic incorrectly switches the second FC active.