ArduPilot/ardupilot #28210: Flash savings ideas : features/drivers to disable/remove to save flash

rmackay9, 2024-09-24:
AP's flash size is constantly growing as new features and drivers are added but unfortunately the STM32 autopilot's CPUs flash size is not keeping up.  This means we need to disable or remove lesser used features from many boards.

This issue is to collect ideas (primarily from members of the core dev team) on features/drivers they think could be sacrificed.

Note we have a few ways to reduce flash:

1. move the feature/driver to be disabled by default thus forcing users to rely on the [custom build server](https://custom.ardupilot.org/)
2. migrate the feature to Lua forcing the user to load a Lua script
3. completely remove the feature or driver

Ryanf55, 2024-09-24:
* AP_GPS_GSOF once we have it available as an EAHRS.

timtuxworth, 2024-09-24:
Plane: suggesting all of these could be custom build server options only.
ICE
Precision Landing
Gimbals (except for Mount and Servo Gimbal)
VTX
Generator

amilcarlucas, 2024-09-24:
Winch
Exotic ESCs

peterbarker, 2024-09-25:
I've used some of these ideas in https://github.com/ArduPilot/ardupilot/pull/28226/files - thanks!

peterbarker, 2024-10-26:
- [ ] bootloader flashing

Hwurzburg, 2025-03-26:
-[] Rangefinder on Plane

tridge, 2025-03-26:
@andyp1per suggested -Os or -O1 default on current 2M boards

peterbarker, 2025-03-27:
[disable-in-turn.csv](https://github.com/user-attachments/files/19495856/disable-in-turn.csv)

That's the output from `./Tools/autotest/test_build_options.py`'s "disable in turn" step.  It shows the number of bytes saved on CubeOrange when each feature is disabled.  This might help selection of things to boot from the standard build.

rmackay9, 2025-03-28:
Thanks @peterbarker.  I've extracted your list and sorted it for copter

<html>
<head>
</head>

<body>

Feature | copter
-- | --
AP_SCRIPTING_ENABLED | 182160
HAL_NAVEKF3_AVAILABLE | 171400
HAL_ENABLE_DRONECAN_DRIVERS | 96120
HAL_LOGGING_ENABLED | 95704
HAL_MOUNT_ENABLED | 67300
AP_FENCE_ENABLED | 50360
AP_RANGEFINDER_ENABLED | 48056
HAL_GYROFFT_ENABLED | 42308
OSD_ENABLED | 40568
AP_RCPROTOCOL_ENABLED | 35888
AP_OPTICALFLOW_ENABLED | 33816
AP_FILESYSTEM_ROMFS_ENABLED | 32912
HAL_MSP_ENABLED | 31808
AP_BOOTLOADER_FLASHING_ENABLED | 31648
HAL_EXTERNAL_AHRS_ENABLED | 29112
HAL_PROXIMITY_ENABLED | 26824
HAL_ADSB_ENABLED | 25800
AP_CAMERA_ENABLED | 25192
AP_FRSKY_TELEM_ENABLED | 24288
AP_AIRSPEED_ENABLED | 22984
AP_FRSKY_SPORT_TELEM_ENABLED | 21952
AP_OAPATHPLANNER_ENABLED | 19016
HAL_WITH_MSP_DISPLAYPORT | 18248
AP_FRSKY_SPORT_PASSTHROUGH_ENABLED | 18240
HAL_SOLO_GIMBAL_ENABLED | 17192
AP_AVOIDANCE_ENABLED | 15680
HAL_EFI_ENABLED | 14816
COMPASS_CAL_ENABLED | 14592
AP_BEACON_ENABLED | 14424
EK3_FEATURE_OPTFLOW_FUSION | 12816
AP_GPS_UBLOX_ENABLED | 12368
AP_RCPROTOCOL_CRSF_ENABLED | 11640
OSD_PARAM_ENABLED | 11596
AP_BARO_PROBE_EXTERNAL_I2C_BUSES | 11296
HAL_VISUALODOM_ENABLED | 10520
AP_EXTERNAL_AHRS_INERTIALLABS_ENABLED | 10344
AC_PRECLAND_ENABLED | 10144
AP_INERTIALSENSOR_HARMONICNOTCH_ENABLED | 9824
HAL_CRSF_TELEM_ENABLED | 9680
AP_VIDEOTX_ENABLED | 9036
HAL_MOUNT_XACTI_ENABLED | 8968
HAL_PICCOLO_CAN_ENABLE | 8920
AP_RCPROTOCOL_SRXL2_ENABLED | 8584
HAL_GENERATOR_ENABLED | 8544
HAL_ADSB_SAGETECH_MXS_ENABLED | 8376
HAL_INS_TEMPERATURE_CAL_ENABLE | 8040
HAL_MOUNT_TOPOTEK_ENABLED | 7092
AP_GPS_NMEA_ENABLED | 6176
HAL_MOUNT_SIYI_ENABLED | 5992
AP_CAMERA_RUNCAM_ENABLED | 5600
AP_OPTICALFLOW_PIXART_ENABLED | 5536
AP_RPM_ENABLED | 5504
AP_WINCH_ENABLED | 5060
GPS_MOVING_BASELINE | 5040
AP_NOTIFY_TONEALARM_ENABLED | 4984
EK3_FEATURE_DRAG_FUSION | 4984
AP_GPS_SBF_ENABLED | 4904
AP_GPS_SBP_ENABLED | 4864
AP_SERVO_TELEM_ENABLED | 4864
AP_EXTERNAL_AHRS_VECTORNAV_ENABLED | 4816
[... 268 lines cut ...]
AP_PLANE_BLACKBOX_LOGGING | -24
AP_PLANE_GLIDER_PULLUP_ENABLED | -24
AP_PLANE_SYSTEMID_ENABLED | -24
AP_PROXIMITY_HEXSOONRADAR_ENABLED | -24
AP_PROXIMITY_MR72_ENABLED | -24
AP_ROVER_ADVANCED_FAILSAFE_ENABLED | -24
AP_SERIALMANAGER_IMUOUT_ENABLED | -24
AP_TEMPERATURE_SENSOR_MCP9600_ENABLED | -24
AP_TEMPERATURE_SENSOR_SHT3X_ENABLED | -24
AP_TEMPERATURE_SENSOR_TSYS01_ENABLED | -24
AP_TEMPERATURE_SENSOR_TSYS03_ENABLED | -24
FORCE_APJ_DEFAULT_PARAMETERS | -24
HAL_NAVEKF2_AVAILABLE | -24
AC_POLYFENCE_FENCE_POINT_PROTOCOL_SUPPORT | -32
AP_DRONECAN_VOLZ_FEEDBACK_ENABLED | -32
AP_MAVLINK_MAV_CMD_SET_HAGL_ENABLED | -32
AP_MAVLINK_RALLY_POINT_PROTOCOL_ENABLED | -32
AP_OSD_LINK_STATS_EXTENSIONS_ENABLED | -32
AP_PLANE_OFFBOARD_GUIDED_SLEW_ENABLED | -32
AP_TEMPERATURE_SENSOR_MLX90614_ENABLED | -32
HAL_PERIPH_SUPPORT_LONG_CAN_PRINTF | -32
HAL_QUADPLANE_ENABLED | -32
AP_AIRSPEED_NMEA_ENABLED | -56
</body>
</html>

amilcarlucas, 2025-05-30:
It would be great if this information would find it's way into custom.ardupilot.org 
The users would be able to get a feeling for how "expensive" each function is.

timtuxworth, 2025-05-30:
How much flash could we save by moving English language text messages (mostly sendtext) out to the GCS?

Hwurzburg, 2025-05-30:
[disable-in-turn.pdf](https://github.com/user-attachments/files/20520924/disable-in-turn.pdf)
I was going to add this to the wiki, so I converted all NEGATIVEs to -na- but there are many that show as na across the boared..
@peterbarker whats the explaination?