ArduPilot/ardupilot #28223: Can't control VTX power on SpeedyBeeF405Wing Plane

IvanLeu, 2024-09-24:
I can't control vtx power using my SpeedyBeeF405Wing, even though VTX, Tramp and SmartAudio features are added by custom build firmware. Tried various of configurations, the power just isn't changing
[image]
[image]
Tried on arduplane 4.5.5 and 4.5.6

IvanLeu, 2024-09-24:
Also in mission planner there's no description for VTX_ENABLE params and other vtx related params, so the only way to set them is to go to params documentation

setupfpv, 2024-09-30:
BAUD need to be "4". Read documentation.

IvanLeu, 2024-09-30:
The problem was that you need to reboot the controller from changes to apply. That helped me