ArduPilot/ardupilot #28242: EKf3: control scaling while using Optical Flow leads to drift in high winds

rmackay9, 2024-09-26:
When the EKF uses optical flow it provides a control scaler which is used by the position controller to limit the velocity feed-forward and acceleration.  This works OK in calm conditions but if the wind is strong enough the control outputs are not enough to allow the vehicle to control its position and it drifts away

This can be reproduced in SITL by doing the following:

- use sim_vehicle.py to start a Copter
- load the [copter-optflow.parm](https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/default_params/copter-optflow.parm) file and restart SITL ([some instructions on the wiki here](https://ardupilot.org/dev/docs/adding_simulated_devices.html#adding-an-optical-flow-sensor))
- param set RNGFND1_MAX_CM 7000
- param set SIM_WIND_SPD 10
- arm and takeoff to 50m and note that the vehicle drifts
- map set showsimpos 1  (<-- to prove that the EKF is estimating the position correctly, it's the controller's that can't keep up)

[AC_PositionControl consumes the EKF3 control limits here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_PosControl.cpp#L640)

This issue is based on this [discussion](https://discuss.ardupilot.org/t/what-is-limiting-the-maximum-lean-angle-for-position-controller-in-non-gps-modes/123329)

satyamk1234, 2026-01-28:
Hi Randy, 
I have tried to reproduce the drift situation in the SITL in strong wind condition. After applying 10m/s wind prevailing west , the copter is commanded to climb to a height of 60 m approx but after 30 m it starts drifting towards west and the EKF failsafe automatically switches to mode LAND. And after switching to mode ALT_HOLD  manually the drift becomes more prominent, the same can be inferred from the OPTICAL_FLOW.flow_comp_m_x graph in the video.

To solve this issue I am currently exploring the first choice given by you and would like to submit a  PR for the same soon.

Simulation Video : https://drive.google.com/file/d/1GxQHdEEhM4iYPVFY7WqHmUUzSyayOiI3/view

rmackay9, 2026-01-28:
Hi @satyamk1234, make the rangefinder range needs to be increased..