ArduPilot/ardupilot #28266: Support for RP2350

akhodeir, 2024-09-30:
## Feature request

**Is your feature request related to a problem? Please describe.**

Raspberry PI announced RP2350 which is more powerful than RP2040
it has:

- Dual Arm Cortex-M33 cores with hardware single-precision floating point and DSP instructions @ 150MHz.
- 520 KB on-chip SRAM
- 30 or 48 GPIO
- ... More
- it is much cheaper than equivalent STM32 microcontrollers.

**Describe the solution you'd like**
I wonder if there is any activity to support RP2350 in near future.

**Describe alternatives you've considered**
N/A

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

Reference : https://www.raspberrypi.com/products/rp2350/

vidma, 2024-12-17:
FYI @akhodeir  it seems there's (unofficial) pi pico support in PX4 (probably RP2040 https://github.com/PX4/PX4-Autopilot/pull/18083 ). I'd still prefer Ardupilot at the moment.

https://github.com/PX4/PX4-Autopilot/tree/main/boards/raspberrypi/pico

also https://github.com/vxj9800/openFC2040 might be nice to check and adapt this opensource PCB for RP2350 if needed. seems  rather clean with BOM etc for easyEDA.
https://www.youtube.com/watch?v=W5kxMzw6kqA
a quick check  with chatgpt o1 says it's pin compatible: https://chatgpt.com/share/e/67617f09-e9b4-8013-a998-9a13e9517108

khancyr, 2024-12-31:
we got nothing to do with how PX4 support pi pico. 
ArduPilot is mainly using Chibios as RTOS, so until there is an official version of chibios on pi pico, we won't support them. 
Another solution would be somebody to step up and build another HAL to support another RTOS on ardupilot.

vidma, 2025-01-02:
@khancyr at least RP2040 seem to be already official in Chibios: https://www.chibios.org/dokuwiki/doku.php?id=chibios:articles:rt_nil_performance#benchmarks (and already included in current ChibiOS referenced [sub-module commit from ardupilot master in: [modules](https://github.com/ArduPilot/ardupilot/tree/master/modules) ).
so maybe RP2040 is a good place to start? or not worth the effort due to more limited hardware than RP2350?

my point about PX4 was as potential reference/inspiration (not sure if might help at all), like sometimes taking pins or other hw info from inav for some less common f405 boards might help when building a new file for [libraries](https://github.com/ArduPilot/ardupilot/tree/Plane-4.4/libraries)/[AP_HAL_ChibiOS](https://github.com/ArduPilot/ardupilot/tree/Plane-4.4/libraries/AP_HAL_ChibiOS)/[hwdef](https://github.com/ArduPilot/ardupilot/tree/Plane-4.4/libraries/AP_HAL_ChibiOS/hwdef) , but indeed here the difference is possibly larger

kwikius, 2025-08-28:
Some work on a flight controller for RP2350 here . https://github.com/qqqlab/madflight . Not ardupilot but certinly would help in the porting

vidma, 2025-08-28:
> Also uses FreeRTOS. 

What do you mean by "also uses" ? Ardupilot uses ChibiOS. PX4 uses NuttX.

Almost certainly I'd prefer Ardupilot to PX4, however,  as ChibiOS was far more not ready for RP2350, here's some in process work on https://github.com/PX4/PX4-Autopilot/pull/24666  . A lot of things work already (!), but still missing a few major thingies: PWM, potentially fix SPI IMU accel, and maybe some I2C instabilities.
Testing it with [Madflight FC1](https://madflight.com/Board-FC1/) !

In long term I would be interested in helping out with migrating Ardupilot on RP2350, if there's more folks interested.

kwikius, 2025-08-30:
> [@kwikius](https://github.com/kwikius) What do you mean by "also uses" ? Ardupilot uses ChibiOS. PX4 uses NuttX.

"also uses" ,  since I did a port of Ardupilot to FreeRTOS once upon a time, Some info  about it here: https://discuss.ardupilot.org/t/why-not-freertos/106060/10?u=skyscraper , so I have a soft spot for FreeRTOS, but also that port I believe was influential in Ardupilot moving away from Nuttx ( The FC ran a software graphical OSD on the STM32F05 I used. Video in above link) , which happened some few months later: https://www.youtube.com/watch?v=y2KCB0a3xMg&t=2s

Worth watching that Ardupilot Conference video  in relation to your PX4 pr, because Ardupilot tended to have a lot of timing issues under Nuttx/Linux/Unix and the video shows the dramatic improvement by switching to Chibios.  

On the subject of getting Madflight running.. Has it actually flown? Video of Madflight "in flight" seems to be hard to come by!

kwikius, 2025-09-04:
@vidma  I realise the links about my Ardupilot FreeRTOS port go to the wrong place since I moved the original repo. Will have to fix that at some point. Most of the Ardupilot HAL work I did is in the https://github.com/kwikius/aerflite repo in the 
https://github.com/kwikius/aerflite/tree/master/libraries/AP_HAL_Quan directory.

Bear in mind the work was done nearly 10 years ago and was never meant to be generic .

MadFlight sounds like great fun. I would like to try it out on my cat, which uses Ardupilot to "fly a hull" as you can see here https://youtu.be/vUaxtkmmM-4?t=102 .  Would be a nice simple application for Madflight.

kwikius, 2025-09-04:
Oh andthere is an ESP32 HAL which uses FreeRTORS I think https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_ESP32

vidma, 2025-09-04:
> Oh andthere is an ESP32 HAL which uses FreeRTORS I think https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_ESP32

yup, seen it already. I guess the problem is reliance on `esp-idf`. Is it somehow alike to pico-sdk ?
https://github.com/ArduPilot/ardupilot/blob/629c6783d267dee67d996ce2ba20bc3f87eb5912/libraries/AP_HAL_ESP32/targets/esp32/esp-idf/CMakeLists.txt#L7-L31

kwikius, 2025-09-13:
Hi @vidma .  I would love to help with this. I love the RP2350 microcontroller and FreeRTOS. I would also love to make some flight controller hardware for it. Unfortunately though, I have too many other projects already that I don't have enough time for. 

I had a look at the current Ardupilot Chibios and FreeRTOS schedulers and they look backwards to me. 

My understanding of the RTOS scheduler is that you run things that require attention fast and often ( like sensors) in high priority threads, but Ardupilot seems to run the main loop in the (2nd) highest priority thread.  Maybe I have that wrong though. Anyway would be worth reading up on RTOS Scheduler theory as it is the core of everything.  If you get a solid design life is quite easy but if not you can have a lot of problems.  But I suspect if I did get involved I would be on an immediate crash course with the devs!