ArduPilot/ardupilot #28274: GCS position in LUA

twistedwings, 2024-10-01:
Some GCS (at least QGC) send their position via MAVLink GLOBAL_POSITION_INT message.
The idea is to persist the position of the GCS and make it available in LUA.
That would be very handy for many use cases, e.g. quickshots.

IamPete1, 2024-10-01:
You can receive MAVLink in scripting so you could track it "manually".

twistedwings, 2024-10-01:
yeah, that might work.
But seems kinda low level.

twistedwings, 2024-10-18:
I have written a lua script witch parses the GLOBAL_POSITION_INT message.
It works very good.

However it would be so much nicer if ArduPilot did this.
The message is already being parsed so it shouldn't be too much effort to persist the position.

LupusTheCanine, 2024-10-21:
There may be more than one other "vehicle" on the network.

timtuxworth, 2024-10-21:
> There may be more than one other "vehicle" on the network.

Each vehicle/GCS should have a unique SYSID_THISMAV - each MAVlink message identifies the SYSID that it came from.  QGC defaults to sending as SYSID 255, but it can be changed in Application Settings.

twistedwings, 2024-10-22:
Do you mean how to identify the GCS ?
I think the only way is to check compid for MAV_COMP_ID_MISSIONPLANNER (190).
Because the sysid of the GCS could be any value although it is 255 in most cases.

peterbarker, 2026-11-15:
There are a vast number of things ArduPilot could keep around on the off-chance that it would be useful to a Lua script.

We provide facilities and examples to let Lua scripts do this themselves - they should do that.

Closing this now.

closed 2026-11-15 by peterbarker