ArduPilot/ardupilot #28277: weird rc input in gcs only setup

twistedwings, 2024-10-01:
## Bug report

In a  gcs only setup all rc inputs are disabled.
After take off the copter starts turning.

Seems to be fixed in 4.6.0.

**Version**
ArduCopter V4.5.7-beta1 (9a755a82)
MissionPlanner 1.3.82

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X

**Hardware type**
SITL

**Logs**
[2024-10-01 17-54-09.zip](https://github.com/user-attachments/files/17212706/2024-10-01.17-54-09.zip)

peterbarker, 2024-11-24:
I don't think this is an ArduPilot issue.

SITL defaults to having non-zero RC inputs on its inputs - but the RC inputs in this log are 0 across the board.  That seems to say something (probably MissionPlanner) has fed values into ArduPilot via its RC input port, despite your settings.

How that's different between different versions of ArduPilot I do not know.

More information required - I'm unable to replicate locally here with normal-SITL.

twistedwings, 2024-11-24:
In 4.5.7 ArduCopter seems to accept rc overrides although RC_OVERRIDE_TIME and RC_PROTOCOLS are zero.
Seems to be fixed in 4.6.0.