ArduPilot/ardupilot #28288: AP_DDS - Multiple Battery interface

tizianofiorenzani, 2024-10-02:
## Multiple battery monitor interface to AP_DDS

I would like to share information of all the available battery monitors through the DDS interface with my companion PC.

**Proposed solution**
Currently we share one battery instance with the topic  `ap/battery/battery0` of type [`sensor_msgs/msg/BatteryState`](https://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/BatteryState.html).

A simple solution would be to use the same topic to publish all the available instances up to `AP_BATT_MONITOR_MAX_INSTANCES` , using the `frame_id` in the header to report the battery ID.

**Alternatively**
- Rename the topic to `ap/battery/batteries`

**Platform**
[x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Ryanf55, 2024-10-02:
Let's
* Update the topic to be `/ap/battery`
* Use the frame ID to differentiate the instances
* Instead of publishing for each instance in `AP_BATT_MONITOR_MAX_INSTANCES`, I would recommend using AP_BattMonitor::BattMonitor_State::type and publishing when the type is NOT equal to `NONE`.

tizianofiorenzani, 2024-10-02:
@Ryanf55 that seems like a plan, I can work out some software and create a PR so we can discuss on that.

tizianofiorenzani, 2024-10-02:
Here is the PR: https://github.com/ArduPilot/ardupilot/pull/28291

Ryanf55, 2024-10-20:
Closed by #28291

closed 2024-10-20 by Ryanf55