ArduPilot/ardupilot #28292: AP_DDS: ArduPilot Parameter Reading/Writing through DDS interface

PQuill33, 2024-10-02:
It would be helpful to have read/write access to ardupilot parameters for all vehicle types through the ROS2/DDS interface.

**Proposed Solution**
Use ros2 service calls to set/read parameters from ArduPilot by exposing the SetParameters and GetParameters interface from rcl_interfaces.

To set a parameter:
- Create `/ap/set_parameters` service of type `rcl_interfaces/srv/SetParameters.srv`
- Response is an array of type `rcl_interfaces/msg/SetParametersResult.msg`

To retrieve a parameter:
- Create `/ap/get_parameters` service of type `rcl_interfaces/srv/GetParameters.srv`
- Response is an array of type `rcl_interfaces/msg/ParameterValue.msg`

I would add code to the AP_DDS_Client interface to get the desired result with the service calls.

**Alternative Approach**
Is it possible to declare the ardupilot parameters in a way that the ros2 param set/get CLI works? My concern is that there are a lot of parameters and I'm not sure the best way to declare them all for this kind of usage.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
rcl_interfaces: https://github.com/ros2/rcl_interfaces

Similar to issue in ticket: [#23309](https://github.com/ArduPilot/ardupilot/issues/23309)

PQuill33, 2024-10-03:
Here is a draft PR with some work towards a set parameter service
[#28298 ](https://github.com/ArduPilot/ardupilot/pull/28298)

Ryanf55, 2024-11-15:
This is now complete. Woohoo!

closed 2024-11-15 by Ryanf55