ArduPilot/ardupilot #28295: DroneCAN: RC link statistics

olliw42, 2024-10-03:
DroneCAN supports RC receivers via the https://github.com/dronecan/DSDL/blob/master/dronecan/sensors/rc/1140.RCInput.uavcan message. 

However, sadly, it only has a "quality" field which by looking into the AP code is in fact RSSI in units of 0 ... 255. 

Today however most people in the FPV arena also want to get LQ, and possibly SNR, and especially want to put them on their OSD. There is no support yet for this. As much as I know support for extending DroneCAN messages definitions while retaining compatibility with existing implementations is not yet existing or in the making. So, extending the existing RCInput seems not a viable path.

This leads to a pretty ugly situation and massively reduces the usefullness of DroneCAN RC (the situation is actually similarly sad as with the MAVLink receiver link stats not being there). To give a sense of it as I perceive it based on user responses: This makes folks wanting CRSF in addition ... just for the purpose of also getting the link stats. Obviously this is a bummer for using DroneCAN since the promise of simplified wiring cannot be kept and the advantages over normal serial largely evaporates. This therefore makes folks wanting CRSF via DroneCAN serial ... which appears to not work with AP, but if it would, doing such work arounds is hardly in the intention of DroneCAN, and also ugly as one then consumes two of the DroneCAN serial ports.

I have no idea what the "best" path and approach would be to correct the situation, but I feel I should raise it as an issue. 

Three approaches would come to my mind:
1. Add a DroneCAN link statistics message. In view of the slow and difficult process with the similar MAVLink link statistics message this may not look as the best path.
2. Add a revised RCInput version 2 message, with few little more fields for the link stats.
3. Add flags to teh existing RCInput message so that the quality field could host different metrics (like RSSI, LQ, SNR), Thuis would at least allow to send the data round robin, i.e. packet #n send RSSI, #n+1 sends LQ, #n+2 send SNR, and so on

tridge, 2024-10-07:
> 3\. Add flags to teh existing RCInput message so that the quality field could host different metrics (like RSSI, LQ, SNR),

my preference is to add a STATUS flag in the existing message that indicates quality is LQ or SNR

tridge, 2024-10-07:
maybe assign 3 of the status bits as what type the quality is?

peterbarker, 2024-10-07:
@olliw42 tear off three bits and use them to indicate what the RSSI field *actually* represents - that's basically your option (3), I think?

olliw42, 2024-10-07:
ok. Will do that :)
THX!

olliw42, 2024-10-08:
pl see https://github.com/dronecan/DSDL/pull/56