ArduPilot/ardupilot #28297: Scripting: mavlink:send_chan does not send anything

twistedwings, 2024-10-03:
## Bug report

mavlink:send_chan does not send any data.

also reported [here](https://discuss.ardupilot.org/t/mavlink-send-chan-function-doesnt-do-anything/122805)

**Issue details**
can be reproduced with this example:
libraries/AP_Scripting/examples/MAVLink_Commands.lua
No ACK is sent.

**Version**
4.5.6
4.6.0

**Platform**
[  ] All
[  ] AntennaTracker
[X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X

**Hardware type**
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2024-10-03:
I have not tried sending command-long messages but other messages can be sent and I've recently done it with the copter-slung-payload.lua script included in this PR https://github.com/ArduPilot/ardupilot/pull/27783

twistedwings, 2024-10-04:
it works.
Made a new test with a minimal lua.

closed 2024-10-04 by twistedwings