ArduPilot/ardupilot #28299: AP_DDS: Provide current Goal

tizianofiorenzani, 2024-10-03:
## Provide current goal location to the DDS Interface

Providing the current goal would help to verify whether the vehicle is actually going where it has been guided to.

**Proposed Solution**
- Create a new topic `/ap/goal` of type GeoPoseStamped, and use it to send the goal location. 
- The goal location in the DDS client should be recovered with `AP::vehicle()->get_target_location(Location &target_loc);`

**Altitude frame**
- Goal could be either be converted to MSL or we could provide the altitude frame in the `frame_id`.

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Ryanf55, 2024-10-03:
In what situation does a companion computer not know what where the autopilot is headed? Won't all commands come from the companion computer over reliable transport?

tizianofiorenzani, 2024-10-03:
Not really, as the goal is sent by topic, so you trust that the vehicle obeys, which I find it too flimsy.

tizianofiorenzani, 2024-10-10:
Created a PR: https://github.com/ArduPilot/ardupilot/pull/28372

peterbarker, 2025-05-09:
Well, we merged the PR, so closing this one :-)

closed 2025-05-09 by peterbarker