ArduPilot/ardupilot #28300: AP_DDS: Provide airspeed for planes

tizianofiorenzani, 2024-10-03:
## Airspeed for planes

The DDS interface does not provide the airspeed, which is an essential information for planes (subs?).

**Solution**
- Publish the airspeed using the same `/ap/twist/filtered` topic, with a dedicated `frame_id = "airspeed"`.
- Leave the `frame_id` empty for standard Inertial/Body

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[ ? ] Submarine

Ryanf55, 2024-10-04:
Airspeed would be great to add to the interface. Airspeed doesn't have a rotational component though, so twist seems overkill. Did you consider publishing just a  `geometry_msgs/msg/Vector3` with the east-north-up component of estimated wind?

tizianofiorenzani, 2024-10-04:
I like this idea, we could use the three components to publish.

- x: airspeed
- y: angle of attack
- z: sideslip angle

Ryanf55, 2024-10-04:
Sideslip is only for plane. Why not just XYZ relative to body frame.

tizianofiorenzani, 2024-10-04:
You mean airspeed in body frame? That would work. length on vector is what we call "airspeed", which is the x component unless you have AOA and sideslip measurements.

Ryanf55, 2024-10-04:
> You mean airspeed in body frame? That would work. length of vector is what we call "airspeed", which most of the times will result in only the x component unless you have AOA and sideslip measurements, or they are estimated internally.

Yea, that's what I meant.

tizianofiorenzani, 2024-10-04:
I can work this out in a PR and we can move ahead from there.

tizianofiorenzani, 2024-10-05:
Well, from inspecting the code it seems pretty easy to obtain the TAS vector:
```
    Vector3f airspeed_vec_bf;
    if (AP::ahrs().airspeed_vector_true(airspeed_vec_bf)) {
        // we are running the EKF3 wind estimation code which can give
        // us an airspeed estimate
        return airspeed_vec_bf.length();
    }
```

Ryanf55, 2024-10-05:
Awesome, at what rate do you think it should be published?

tizianofiorenzani, 2024-10-05:
Here is the PR: https://github.com/ArduPilot/ardupilot/pull/28319

tizianofiorenzani, 2024-10-06:
> Awesome, at what rate do you think it should be published?

I publish it within the same velocity timer, if airspeed is available.

peterbarker, 2025-05-09:
PR was merged - closing this one.

Rememinder that if you use `Closes #ISSUENUMBER` in your PR then github will magically close these issues as completed when the merge is done.

closed 2025-05-09 by peterbarker