ArduPilot/ardupilot #28303: Feature request: Support negative RPM for dynamic harmonic notch

RickReeser, 2024-10-04:
Some specifications allow ESC telemetry to send negative RPM if the motor is spinning backwards (or just clockwise vs. counter-clockwise, as in my case), e.g. DroneCAN esc status message. RPM-tracking notches do not use negative RPM and they will just sit at the minimum frequency:
[image]

Suggestion:
I suppose the naive solution of using the absolute value of RPM would suffice?

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2024-10-04:
Certainly some judicious fabsf() would probably fix this and would be easy to do.

andyp1per, 2024-10-04:
Please can you try

RickReeser, 2024-10-04:
Yeah, I will look into it today. I don't have any familiarity with the filtering classes, but I suppose this will be a good introduction.

andyp1per, 2024-10-04:
> Yeah, I will look into it today. I don't have any familiarity with the filtering classes, but I suppose this will be a good introduction.

Did you see the PR I did?

RickReeser, 2024-10-04:
Nope, just saw it. Will try your PR now.

closed 2024-10-09 by tridge