ArduPilot/ardupilot #28330: MATLAB Simulink ROS2

harunkurtdev, 2024-10-07:
## Feature request

By using Matlab and Simulink, controlling the autopilot using Ardupilot msgs and ROS2.

[image]

**Describe the solution you'd like**
Praitical Control architectures can be applied via Simulink and ros2 package can be obtained. We can also use different control systems.

**Platform**
[X ] All
[X ] All OS

amilcarlucas, 2024-10-07:
Explain what you expect to happen. Explain what is happening instead.
It is not clear what you want. ArduPilot already has ROS2 support.
Why did you mark some topics in yellow?

harunkurtdev, 2024-10-07:
Mr. @amilcarlucas Sir, 

I wanted to draw attention to ardupilot_msgs here.
While developing Ros2 on Ardupilot, I think Matlab Simulink support can be added.

During our student years, we communicated with PX4 and ROS2 and did some work on Simulink.
However, once Ardupilot is stable, a documentation can be written on this subject.

harunkurtdev, 2024-10-07:
@amilcarlucas 

If you want, I can leave a simple example, ROS2 with Simulink

srmainwaring, 2024-10-07:
Hello @harunkurtdev, I'm afraid it's still not clear what feature you're asking for.

Are you requesting that we write a matlab module that uses the ROS 2 interface exposed by AP_DDS?

harunkurtdev, 2024-10-07:
Mr. @srmainwaring, 

I may have misunderstood. But what I want is this;

I think a guide explaining "how to connect ardupilot to matlab with ros2" would be useful. I can write this document. Where should I write it?

srmainwaring, 2024-10-07:
> Where should I write it?

Additional documentation and guides are always welcome! To start with you could write an article in ArduPilot Discourse which could be submitted as a blog. The follow up would be to add a page to the Dev wiki. There is a section for ROS 2: https://ardupilot.org/dev/docs/ros2.html