ArduPilot/ardupilot #28340: AP_DDS: Takeoff interface for Copter/QPLANE

tizianofiorenzani, 2024-10-07:
## Takeoff Service interface

I would like a new service that allows to command takeoff for copters/qplane. Current interfaces lack the ability to takeoff a copter, as commanding a location in Guided mode fails, unless vehicle is flying.

**Solution**
- Service `/ap/start_takeoff` with argument `altitude` = takeoff altitude in meter.
- We can use the interface `AP::vehicle()->start_takeoff(alt)`, which returns false by default, if not overwritten by the vehicle implementation.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

Ryanf55, 2024-10-08:
See https://github.com/ArduPilot/ardupilot/pull/26911

tizianofiorenzani, 2024-10-08:
@Ryanf55 that was easy ;-). Closing