ArduPilot/ardupilot #28350: ArduCopter - Manual Control Input (Joysticks Input) Are Ignored When Sent from MAVSDK

rayw-dronesense, 2024-10-08:
## Bug report

**Issue details**

Manual control input sent from MAVSDK are ignored by ArduPilot while they work on PX4.

Specifically this API,

https://github.com/mavlink/MAVSDK/blob/v2.12.9/src/mavsdk/plugins/manual_control/manual_control_impl.cpp#L92

Our investigation found that MAVSDK uses "system id" 245 as the ground control station, but for some reason ArduPilot is expecting "system id" to be  255 and everything else is dropped. Would there be any specific reason why ArduPilot would be looking for a hardcoded ID?

**Version**
v4.4.0, v4.6.0 all observed this issue - in simulation (SITL) and real aircraft

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

Quad sim (SITL) and Hexa (real aircraft)

**Hardware type**

For sim it's not applicable. For real aircraft, it's a proprietary system based on ArduPilot which behaves identically to the SITL simulator.

**Logs**

There are no logs - the inputs sent are ignored silently.

rmackay9, 2024-10-09:
Hi @rayw-dronesense,

AP has a [SYSID_MYGCS](https://ardupilot.org/copter/docs/parameters.html#sysid-mygcs-my-ground-station-number) parameter that should allow changing the sysid that is accepted.  Could you give that a try?

There's also a [SYSID_ENFORCE](https://ardupilot.org/copter/docs/parameters.html#sysid-enforce-gcs-sysid-enforcement) parameter that controls whether that is the only SYSID that is accepted for commands and RC controls.

rayw-dronesense, 2024-10-09:
@rmackay9 Thanks for getting back to me. Even if that works, that would make it inconvenient for the user of a piloting app that may at some point switch back to using QGroundControl, right?

Is there any particular reason the software stack is doing these ID checks and rejecting all other IDs?

https://github.com/ArduPilot/ardupilot/blob/9d6190b4fd77f28fe56851fa87f8676fb80ad1d8/libraries/GCS_MAVLink/GCS_Common.cpp#L7070

rmackay9, 2024-10-10:
@rayw-dronesense,

I think that unless SYSID_ENFORCE is set to 1 it will accept manual input from any sysid.  Maybe you could give it a try?  The above comment from me is really just a guess so I think a bit more investigation would help clarify if this is really the issue.  It may be that MAVSDK is using a different message that AP doesn't support.

peterbarker, 2024-11-24:
> @rayw-dronesense,
>
> I think that unless SYSID_ENFORCE is set to 1 it will accept manual input from any sysid. Maybe you could give it a try? The above comment from me is really just a guess so I [...]

We enforce source-system-is-SYSID_MYGCS on RC_CHANNELS_OVERRIDE even when SYSID_ENFORCE is not set.

We can change that, but it is long-standing behaviour, so we would want to be sure.

rmackay9, 2024-11-24:
I think we can close this.  @rayw-dronesense if you find it doesn't work just comment here because we will see replies

closed 2024-11-24 by rmackay9

rayw-dronesense, 2024-11-25:
@rmackay9 / @peterbarker  - I just tried `SYSID_ENFORCE` to 0 and unfortunately it was still ignoring the joystick inputs. 

Would you mind explaining why this behavior is like this? It seems PX4 does not do this.