ArduPilot/ardupilot #28363: VRBrain 5.2 and Ardupilot V4.5.6 - CRFS: broken functionality with Yaapu.

robustini, 2024-10-09:
Same parameters, same configuration and version of ELRS (V3.5.1, the latest).
With Ardupilot V4.5.6 Yaapu doesn't interpret CRFS and doesn't work, with Ardupilot V4.4.0 it does, and I don't think it's Yaapu's fault.
Many CRFS inherent commits so I really don't know where I can look to figure out what is broken.
Ping @yaapu...

Ardupilot V4.4.0
[image]

Ardupilot V4.5.6
[image]

tpwrules, 2024-10-09:
For what it's worth, telemetry works fine for me with ELRS 3.3.0, Copter 4.5.4, and Yaapu 1.9.5 on OpenTX 2.3.15. I know the non-Ardupilot stuff is pretty out of date but that might provide a clue.

robustini, 2024-10-10:
> For what it's worth, telemetry works fine for me with ELRS 3.3.0, Copter 4.5.4, and Yaapu 1.9.5 on OpenTX 2.3.15. I know the non-Ardupilot stuff is pretty out of date but that might provide a clue.

Thank you for the feedback.
Everything is up to date here, Ardupilot to the latest release, EdgeTX also, the problem occurs only with VRBRain V5.2, I have other FCs (Cube Orange, Speedybee, CUAV) where this thing does not happen.

robustini, 2024-11-10:
Found the problem and don't understand this change in the hwdef of this board.
On 4.5.0 hwdef for this board:
include ../include/minimize_features.inc

Inside minimize_features.inc:

# Crossfire telemetry must be explicitly enabled on minimized boards:
define HAL_CRSF_TELEM_ENABLED 0

robustini, 2024-11-10:
Found the problem and don't understand this change in the hwdef of this board.
On 4.5.0 hwdef for this board:

include ../include/minimize_features.inc

Inside minimize_features.inc:

define HAL_CRSF_TELEM_ENABLED 0

Compiled by enabling and problem solved.

closed 2024-11-10 by robustini

andyp1per, 2024-11-11:
it should probably be changed to minimize_fpv_osd.inc