ArduPilot/ardupilot #28374: Potential thread stack overflow in AP_SmartAudio::loop

mirusu400, 2024-10-10:
## Bug report

**Issue details**

While some detailed build configuration, There are potential stack overflow in thread function named `AP_SmartAudio::loop`.

https://github.com/ArduPilot/ardupilot/blob/b36f539c7c9db63514882fcf81bac0f00d58574f/libraries/AP_VideoTX/AP_SmartAudio.cpp#L52-L70

In this line, `AP_SmartAudio::loop` Allows 768 stack size.

However, after manually checking stack, it might have 1040 Bytes in with some detailed configurations.

**Steps to produce**

1. In `boards.py`, Add this line:
```diff
# In 224 line..
        env.CFLAGS += [
++          '-fstack-usage',
            '-ffunction-sections',
            '-fdata-sections',
            '-fsigned-char',
...
# In 343 line...
        env.CXXFLAGS += [
++          '-fstack-usage',
            '-std=gnu++11',

            '-fdata-sections',
```

This will help you automatically calculating stack size of each function.

2. Build ardupilot normally, using this configurations:
```
./waf configure --board AnyleafH7 -g --check-verbose --disable-Werror --toolchain=arm-none-eabi --notests --enable-scripting --enable-opendroneid --enable-check-firmware --enable-custom-controller --enable-gps-logging --enable-header-checks --enable-stats 
./waf -v rover
```
Now we can get stack usage file (*.su) for each source file, So we can manually check stack size of each function.

**In case of `AP_SmartAudio::loop`:**

There are large call stack with this flow:
```
AP_SmartAudio::loop (_ZN13AP_SmartAudio4loopEv) => 64 size
AP_VideoTX::announce_vtx_settings (_ZNK10AP_VideoTX21announce_vtx_settingsEv) => 32 size
GCS::send_text (_ZN3GCS9send_textE12MAV_SEVERITYPKcz) => 0 size
GCS::send_textv (_ZN3GCS10send_textvE12MAV_SEVERITYPKcSt9__va_listh) => 160 size
GCS::service_statustext (_ZN3GCS18service_statustextEv) => 24 size
GCS_MAVLINK::service_statustext (_ZN11GCS_MAVLINK18service_statustextEv) => 120 size
_mav_finalize_message_chan_send (_mav_finalize_message_chan_send) => 88 size
mavlink_sign_packet (_Z19mavlink_sign_packetP17__mavlink_signingPhPKhhS3_hS3_) => 136 size
mavlink_sha256_update (_Z21mavlink_sha256_updateP18mavlink_sha256_ctxPKvm) => 416 size

SUM => 1040 bytes
```
So, there are potentially occur stack overflow in `AP_SmartAudio::loop` Thread function.

**Version**

Commit 0e72fc7bd28f5ee5e319e4807bc05a4bf9cbc224

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[x] Rover
[  ] Submarine

I've only checked `Rover` yet, but I think all platforms have same issues.

**Hardware type**
AnyleafH7

tridge, 2024-10-14:
nice find, thanks!
we could either allocate the sha256 context for the mavlink signing separately, or you could just increase the stack here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_VideoTX/AP_SmartAudio.cpp#L59

mirusu400, 2024-10-15:
@tridge Sure! is just changing stack size is enough?

peterbarker, 2024-10-19:
On Tue, 15 Oct 2024, mirusu400 wrote:

> @tridge Sure! is just changing stack size is enough?

Problem with just increasing stack size is that we have a *lot* of 
threads, and adding 512 bytes to each of them is a significant hit to 
resources on smt32.

timtuxworth, 2024-10-19:
> nice find, thanks! we could either allocate the sha256 context for the mavlink signing separately

mavlink_sha256_update uses 416 bytes on the stack!  mavlink_sha256_ctx only seems to be 90 bytes and "zero" seems to be 72 bytes. Where is the rest?

peterbarker, 2024-10-19:
On Fri, 18 Oct 2024, Tim Tuxworth wrote:

>       nice find, thanks! we could either allocate the sha256 context for the mavlink signing separately
> 
> mavlink_sha256_update uses 416 bytes on the stack! mavlink_sha256_ctx only seems to be 90 bytes and "zero" seems to be 72 bytes. Where is the rest?

You know there's only 768 bytes allowed, right?

... and the information *is* there :-)

timtuxworth, 2024-10-19:
> On Fri, 18 Oct 2024, Tim Tuxworth wrote: nice find, thanks! we could either allocate the sha256 context for the mavlink signing separately mavlink_sha256_update uses 416 bytes on the stack! [...]

That's kind of my point. If it's correct then one single method is consuming 416 bytes out of 768 bytes of stack space. I'm saying this is what should be fixed.