ArduPilot/ardupilot #28416: unknown usb device after building new bootloader

superquala, 2024-10-16:
Hello
I cloned ardupilot repository to make some changes. I have no problem with building FC firmwares. the problem is when I build a new bootloader, let me explain
I use a board with stm32h743 mcu.
I use WSL(ubuntu) terminal on my windows 10 laptop
There is no problem when I use the pre build bootloader files in path: ardupilot\Tools\bootloaders
I can upload them to my FC using stm32cubeprogrammer and my laptop(in device manager) recognize it, even if I upload another stm32h743 bootloader
[image]

But when I build a new bootloader with this command: Tools/scripts/build_bootloaders.py board-name
[image]

and upload it with the same way, my laptop(and also my other pc with windows 7) can't recognize it! why?

[image]

I tried it with another FC and I built bootloader from different boards but nothing changed!

peterbarker, 2024-11-24:
Does the bootloader actually work?

Which compiler are you using?

peterbarker, 2025-05-11:
No answer; please re-open if you want to pursue.

closed 2025-05-11 by peterbarker