ArduPilot/ardupilot #28457: RTCM input on serial port.

geofrancis, 2024-10-23:
## Feature request
allow RTCM injection directly to a canbus/serial port without it needing to be encapsulated in mavlink, my understanding is that the rtcm data is already sent intact over mavlink so its just the mavlink decoding that needs to be skipped. 

https://discuss.ardupilot.org/t/rtk-without-mavlink-via-gps2-port/125131

**Platform**
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

khancyr, 2024-10-24:
Shouldn't be hard, downside on this is that you hold a serial port only for injection. At this point, that is probably better to inject directly to GNSS.
But why not

geofrancis, 2024-10-24:
> Shouldn't be hard, downside on this is that you hold a serial port only for injection. At this point, that is probably better to inject directly to GNSS. But why not

The problem is trying to do that with a CAN GPS or a M8P like here + that doesn't have a second serial port.