ArduPilot/ardupilot #28476: BL_HELI: Reverse throttle by RC Switch spins the motor in the wrong directions

menschel, 2024-10-26:
## Bug report
Reporting a minor issue with motor direction with 3D enabled BL_HELI compatible ESCs. Observed with Bluejay and AM32 firmware, both DSHOT600 protocol. Can be fixed by changing ESC to "3D (forward/reverse) **Reversed**".

**Issue details**
While following the guide for Reverse Throttle via RC_Switch
https://notes.stavros.io/ardupilot/reverse-thrust/
I observed that the resulting directions are wrong.
After setting the ESC to 3D (forward/reverse) via ESC-Configurator Online tool and changing the parameters according to guide,
the default motor direction produces reverse thrust. After switching to Reverse Thrust followed by the GCS message that Reverse Thrust is enabled, the motor direction produces positive thrust.

**Version**
Ardupilot 4.5.7

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
* AR Wing Pro (Pusher Type)
* ZOHD Drift Glider (Customized <250g build) (Pusher Type)

**Hardware type**
STM32F405 based FC with 
* Holybro Tekko F4 45A ESC (AM32 Firmware)
* Dys 7A BL-HELI-S ESC (Bluejay Firmware)

**Logs**
The issue does not show in logs but is observed visually on the bench.

geofrancis, 2024-10-26:
have you swapped the phase wires on the motor to change its direction?

menschel, 2024-10-26:
> have you swapped the phase wires on the motor to change its direction?

No, I don't swap any cables in the process.

If ESC is configured "normal" direction, motor spins in the correct "forward" direction.
If the ESC is configured "3D (forward/reverse)" and parameters are changed according to guide, the motor spins reverse while Ardupilot treats it as normal direction and vise versa.

I can fix the whole configuration thing if I set the ESC to "3D (forward/reverse) REVERSED". I currently don't have a PWM controlled ESC to test if it works correctly in PWM control and just wrong in DSHOT.

andyp1per, 2024-10-27:
What parameters have you set?

menschel, 2024-10-27:
> What parameters have you set?

My latest config for the Drift 
SpeedyBee F405 Wing Mini + Dys XSD7A
[current.zip](https://github.com/user-attachments/files/17533591/current.zip)

Motor is on Channel1. Additional Changes have been according to guide.
THR_MIN=-30
SERVO_BLH_3DMASK=1
RC5_OPTION=64
USE_REV_THRUST=0
FLTMODE1=0
FLTMODE2=0

ESC set to bidirectional non-inverted via ESC-Configurator tool.

menschel, 2024-10-27:
The ARWingPro (Matek F405 Wing V2 + Holybro Tekko32 F4 45A ESC) still has this test setup.

It works in correct directions with the attached parameter file and ESC set to bi-directional + reversed. The motor direction is also correct if none of the check-boxes is selected, (i.e. not bi-directional and not reversed).
[image]
[20241027_ARWingPro_ReverseThrottleViaRC5Test.zip](https://github.com/user-attachments/files/17534852/20241027_ARWingPro_ReverseThrottleViaRC5Test.zip)

andyp1per, 2024-10-28:
Pretty sure this is a bug I fixed in 4.6, please can you try the latest?

menschel, 2024-10-28:
Sorry, it is the same behavior in 4.6. 
I'm not sure about the scaling either. Hard to tell on the bench without a prop and it seems that ESC rpm is reported unsigned. I'll take it outside, strap it to some fixture and run it under load in both directions when the rain clears up. Will also be a good test for the folding prop, not sure if it actually can work in reverse.
[image]
[00000004.zip](https://github.com/user-attachments/files/17539340/00000004.zip)

menschel, 2024-10-28:
IMO scaling to -30% throttle also not there. I did not go above half stick range because it was truly scary.
Fold Propeller 9/5
[image]
Stock Propeller 8/5
[image]

andyp1per, 2024-10-31:
I think you are getting bitten by SERVO_DSHOT_ESC - you have configured this and so AP will send the dshot commands to reverse, so given that you have already reversed in the configurator the setup will be inverted. Remove the static setting from the configurator

menschel, 2024-11-01:
> I think you are getting bitten by SERVO_DSHOT_ESC - you have configured this and so AP will send the dshot commands to reverse, so given that you have already reversed in the configurator the setup will be inverted. Remove the static setting from the configurator

For the test with AP 4.6, I did not reverse the ESC via Configurator. I'm seeing the same issue as before. If ESC is configured normal, then direction is forward. If ESC is configured bi-directional and not reversed, then direction is backwards without reverse-thrust and vice-versa.

The dshot esc is for propeller efficiency measurements and with AM32 I can't go back to PWM controlled modes anyways.

Can I debug this DSHOT stuff somehow via logs?

andyp1per, 2024-11-05:
> The ARWingPro (Matek F405 Wing V2 + Holybro Tekko32 F4 45A ESC) still has this test setup.
> 
> It works in correct directions with the attached parameter file and ESC set to bi-directional + reversed. The motor direction is also correct if none of the check-boxes is selected, (i.e. not bi-directional and not reversed). ![AM32](https://private-user-images.githubusercontent.com/24752519/380518450-664f11f9-c780-4220-aa88-0e42f0960ec3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4MTI0MTMsIm5iZiI6MTczMDgxMjExMywicGF0aCI6Ii8yNDc1MjUxOS8zODA1MTg0NTAtNjY0ZjExZjktYzc4MC00MjIwLWFhODgtMGU0MmYwOTYwZWMzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA1VDEzMDgzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQwYjA2NzZmOGI4MTg4MjVlY2NiNDViNzc0MWYxZDEzMzFmY2JjMzIyOTA0Y2I5OWM0NWM2NjRiOWFhYTQxNTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.z3qE91LR49rzPfYJyKZv_Epo-AAT1VH0zn2FqVEXW-E) [20241027_ARWingPro_ReverseThrottleViaRC5Test.zip](https://github.com/user-attachments/files/17534852/20241027_ARWingPro_ReverseThrottleViaRC5Test.zip)

So wait, it works on this setup but not the other?

andyp1per, 2024-11-05:
> SpeedyBee F405 Wing Mini

I don't think this supports bdshot, I am not sure what the implication for 3D is. Should work but who knows. Other possibility is that dshot commands are not working properly.

menschel, 2024-11-05:
> So wait, it works on this setup but not the other?

Yes the ARWingPro with the ESC set to bi-directional and reversed has a correct working Reverse Thrust.
So there is one "reverse" too much in the chain. I assume it's a split throttle range in general and some reverse thrust command working against each other, so one reverse comes from the throttle range and the other likely comes from some command to switch directions.

> I don't think this supports bdshot, I am not sure what the implication for 3D is. Should work but who knows. Other possibility is that dshot commands are not working properly.

Observation tells, no implications, it just does not evaluate the ESC telemetry.