ArduPilot/ardupilot #28497: AP_DDS: Race condition generating DDS messages (microxrceddsgen is not thread safe)

Ryanf55, 2024-10-30:
## Bug report

**Issue details**

Very occasionally, microxrceddsgen fails to generate messages. This is likely because waf is writing to the same file from multiple threads.

**Version**
4.5
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
all
**Hardware type**
sitl
**Logs**

**Details**

microxrceddsgen can take an argument for multiple files at the same time. We call `bld()` once for each file. It would be faster to do batch them per folder, or all together at the same time.

```
microxrceddsgen -cs -replace -default-container-prealloc-size 8 -d build/IDL -I libraries/AP_DDS/Idl/ /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/ardupilot_msgs/msg/GlobalPosition.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/ardupilot_msgs/srv/ArmMotors.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/ardupilot_msgs/srv/ModeSwitch.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/builtin_interfaces/msg/Duration.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/builtin_interfaces/msg/Time.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geographic_msgs/msg/GeoPoint.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geographic_msgs/msg/GeoPointStamped.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geographic_msgs/msg/GeoPose.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geographic_msgs/msg/GeoPoseStamped.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Point.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Pose.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/PoseStamped.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Quaternion.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Transform.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/TransformStamped.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Twist.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/TwistStamped.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Vector3.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/geometry_msgs/msg/Vector3Stamped.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/rosgraph_msgs/msg/Clock.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/sensor_msgs/msg/BatteryState.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/sensor_msgs/msg/Imu.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/sensor_msgs/msg/Joy.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/sensor_msgs/msg/NavSatFix.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/sensor_msgs/msg/NavSatStatus.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/std_msgs/msg/Header.idl /home/ryan/Dev/ardu_ws/src/ardupilot/libraries/AP_DDS/Idl/tf2_msgs/msg/TFMessage.idl
```

We'll need to generate them into the right folder too. See https://github.com/eProsima/Micro-XRCE-DDS-Gen/issues/80

AlSharji0, 2025-03-15:
Hi, 
I'd like to tackle this.
Is anyone working on this?

Ryanf55, 2025-03-15:
No, if you can figure it out on the branch I had started, that would be great. I was pretty close.

AlSharji0, 2025-03-16:
Alright, thanks

peterbarker, 2025-04-29:
@AlSharji0 this bug causes a bunch of CI tests to fail, which is very annoying.  Are you still chasing this?

AlSharji0, 2025-04-29:
@peterbarker I've been on and off this for a week and change, I can't seem to get it to work as intended so far.
If you'd like to take over I wouldn't mind.

srmainwaring, 2025-06-17:
@Ryanf55 reworked version of your patch here: https://github.com/srmainwaring/ardupilot/tree/prs/28497-fix-waf-dds-gen-race-reworked

Seems to be working, a little ugly in places.

closed 2025-06-18 by peterbarker