ArduPilot/ardupilot #28503: CRC32 algorithm for FTP is different from Mavlink docs

BR7RKR, 2024-10-30:
## Bug report

**CRC32 algorithm for FTP is different from Mavlink docs**

CRC32 algorithm for FTP is different from Mavlink docs. Docs state that you should use  polygon 0x04C11DB7 and Start at 0 instead of 0xFFFFFFFF, Missing final XOR out operation with 0xFFFFFFFF.
Here is the link to the mavlink docs: https://mavlink.io/en/guide/crc.html
**latest**

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

tpwrules, 2024-11-01:
These are the same CRC, just expressed using the bit-reversed form of the polynomial (not polygon).

Can you provide further evidence that there is something wrong here? Is some software not compatible?

olliw42, 2024-11-01:
the "Gary Brown" label in fact was/is missleading, the crc32 situation was in depth analyzed and corrected&aligned some time back:

https://github.com/mavlink/mavlink/issues/1208
https://github.com/mavlink/mavlink-devguide/pull/258
https://github.com/mavlink/mavlink-devguide/pull/293

tpwrules, 2024-11-01:
Yes, I agree the label is not conclusive proof and we do deviate from "Gary Brown" as documented in the MAVLink spec. I also agree that I don't see any inconsistency.

If you have a documented incompatibility I would like to see it, otherwise I will close this.

BR7RKR, 2024-11-01:
I think you are right, thanks! However, will the result of an algorithm with a non bit-reversed form of the polynomial be different from this algo?