ArduPilot/ardupilot #28525: HERE4C: The connector board has I2C, but it cannot be used.

muramura, 2024-11-05:
## Bug report

**Issue details**

The connector board has I2C, but it cannot be used.
We cannot obtain voltage and current values from an I2C battery monitor (e.g., INAxxx). For example, the PM03D Power Module.
[image]

UART connector in I2C
[image]

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/Here4FC/README.md

I2C is not included in the features.

**Version**

AC4.5.7

**Platform**

[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

QUAD/CW-X

**Hardware type**

HERE4

**Logs**

NONE

MallikarjunSE, 2024-11-05:
Needs pull up on those pins. Doubtful it has pull up since it's primarily meant for GNSS system.

muramura, 2024-11-06:
@bugobliterator san

Is it acceptable to specify the internal pull-up in the hardware definition file?  
What should I do?  
If possible, I would prefer not to handle it externally.

peterbarker, 2025-04-29:
@muramura did you progress with this?  Seems to be bad if you can't use i2c-in on Here4FC!

... of course, CAN will almost certainly work for you there...

rmackay9, 2025-04-29:
@bugobliterator can probably answer but I think that maybe on that little breakout board that comes with the Here4 the I2C pins are not connected.