ArduPilot/ardupilot #28550: When using ArduPilot connected via MAVROS and increasing the IMU rate using MAV_CMD_SET_MESSAGE_INTERVAL, there's an issue where the rate temporarily returns to the default low rate

engcang, 2024-11-08:
## Bug report

I'm using the following Python code to set the data rates with `MAV_CMD_SET_MESSAGE_INTERVAL`. 
After `MAVROS` connects and the `/mavros/state` topic is received, the code waits for 10 seconds before issuing the `MAV_CMD_SET_MESSAGE_INTERVAL` command.
In MissionPlanner/QGroundControl, all `SRx_EXTRAx` parameters are set to 0.

Right after `MAVROS` connects, the IMU rate is 25Hz. 
1. When the Python code runs, the IMU rate does not immediately increase; it takes a few seconds before it reaches 100Hz.
2. After a few more seconds, the IMU rate drops back to the default 25Hz.
3. Then, after another few seconds, it goes back up to 100Hz and stays there consistently.

What could be causing the rate not to stay at 100Hz right away, as in step 3, without going through steps 1 and 2? 
Could it be that ArduPilot takes longer to finish booting than expected? 
Does it still take additional time even after receiving `/mavros/state`?

```python
#!/usr/bin/env python3
import rospy
from std_msgs.msg import Empty
from mavros_msgs.msg import State
from mavros_msgs.srv import CommandLong, CommandLongRequest

def set_message_interval(message_id, interval):
    try:
        set_interval = rospy.ServiceProxy('/mavros/cmd/command', CommandLong)
        req = CommandLongRequest()
        req.broadcast = False
        req.command = 511        # MAV_CMD_SET_MESSAGE_INTERVAL
        req.param1 = message_id  # Message ID
        req.param2 = interval    # Interval in microseconds
        req.confirmation = 0
        req.param7 = 0           # Target system (0 for broadcast to all)

        response = set_interval(req)
        return response.success
    except rospy.ServiceException as e:
        print(f"Service call failed: {e}")  # Using f-string for formatting
        return False

''' class '''
class Robot():
    def __init__(self):
        rospy.init_node('data_stream_speed_setter', anonymous=True)
        self.state_sub = rospy.Subscriber('/mavros/state', State, self.state_cb)
        self.switch_flag = False
        self.first_data_time = 0

    def state_cb(self, msg):
        if self.first_data_time == 0:
            self.first_data_time = rospy.get_time()
        if rospy.get_time() - self.first_data_time > 10:
            self.switch_flag = True

if __name__ == "__main__":
    setter_ = Robot()
    
    message_intervals = {
        0: 500000, #0 -> HEARTBEAT
        1: 500000, #1 -> SYS_STATUS, including battery
        27: 10000, #27 -> RAW_IMU
        30: 10000, #30 -> ATTITUDE #note: important for orientation alignment
        # 105: 10000, #105 -> HIGHRES_IMU
        32: 33333, #32 -> LOCAL_POSITION_NED
        35: 500000, #35 -> RC_CHANNELS_RAW
        # 33: 33333, #33 -> GLOBAL_POSITION_INT
        # 84: 20000, #84 -> SET_POSITION_TARGET_LOCAL_NED
        # 82: 10000 #82 -> SET_ATTITUDE_TARGET
        # 102: 50000 #102 -> VISION_POSITION_ESTIMATE
    }
    success_ids = set()

    while not rospy.is_shutdown() and len(success_ids) < len(message_intervals):
        if setter_.switch_flag:
            for message_id, interval in message_intervals.items():
                if message_id not in success_ids:
                    if set_message_interval(message_id, interval):
                        print(f"Message interval set successfully for ID {message_id}")
                        success_ids.add(message_id)
                    else:
                        print(f"Failed to set interval for message ID {message_id}, retrying...")
                rospy.sleep(0.5)
        rospy.sleep(0.5)

    print("All message intervals set successfully")
    print("Done setting message intervals")
    rospy.signal_shutdown("Shutting down the node due to successful operation")
```

**Version**
Copter 4.5.6

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2024-11-08:
Mission Planner, MAVProxy and QGCS periodicaly change the SRx_ rates in the background unless you configure them not to do so.

engcang, 2024-11-08:
@amilcarlucas 
As I wrote in the first comment, in MissionPlanner/QGroundControl, all SRx_EXTRAx parameters are set to 0.
Anyway do I have to set MAVProxy and QGCs to not to do so in some ways else?

amilcarlucas, 2024-11-08:
do I have to set MAVProxy and QGCs to not to do so in some ways else?

Yes, you need to start mavproxy with --SETRAM_RATE=-1 and in QGC there is probably something similar. 0 is not enough.

closed 2024-11-08 by amilcarlucas