ArduPilot/ardupilot #28562: Plane will breach fence in GUIDED "Fly To" command when it could reject the command for convex inclusion fence

Ryanf55, 2024-11-08:
## Bug report

**Issue details**

1. Set FENCE_ENABLE to 1
2. Set FENCE_ACTION to 1 (we care about breaching the fence, it's not just level 0)
3. Configure this fence 
[fence-items.txt](https://github.com/user-attachments/files/17685179/fence-items.txt)

4.  `mode takeoff`
5. `mode guided` once at altitude
6. Submit a "Fly To" waypoint INSIDE the inclusion fence, but where part of the inclusion fence obstructs the line connecting the plane and the point.
7. Observe the plane ACCEPT the command, then fly right into the fence.

**Expected Behavior**
If the plane knows that it cannot reach a guided Fly  To waypoint without breaching the fence, reject the command. In the future, invoke a planner and fly around the fence. 

**Actual Behavior**
The `DO_POSITION` command is accepted, the plane flies towards the goal, hits the fence, then recognizes that, switches to RTL, and flies home. 

**Version**
master, hash `963095978b57e976223e58b922e1c415d8bc2090`, ~4.6

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
N/A

**Logs**

Hwurzburg, 2024-11-09:
Missions act the same.....they will breach and take breach action....would it be better to avoid in Guided fly to's...yes...but also we should make prearm fails for missions that fly thru fences....both are generally bad things....copters are slow enough to avoid something more difficult in planes so having improved safeguard to prevent flights or fly-tos into fences would be a good option...the question is how difficult and how much flash?

IamPete1, 2024-11-09:
Because we don't cross track on guided commands you can't assume that we just fly in a straight line, so its not as easy as it might seem. You would have to build a path that takes into account the initial turn. 

The other option of course is not to breach the fence in the first place. Bendy ruler and Dijkstra's will come to plane eventually.

LupusTheCanine, 2024-11-12:
1. ArduPlane not being proactive about fences is a well documented state of things.
2. We shouldn't provide unreliable checks to the user and given effects of wind, turn radius and possible user inputs using stick mixing such early check will be fairly unreliable.
3. We should account for wind effects in proactive fence avoidance as it can have significant effect on effective "turning radius"
4. IMHO Bendy ruler should work but Dijkstra's isn't ready for turn radius considerations of plane.

peterbarker, 2024-11-23:
So the consensus seems to be that we can't tell ahead of time that we will or won't breach a fence.

When we get path-finding (and the user enables it) we could potentially reject targets if the path-finder can't find a path, but in the meantime I think we should close this one - it's well-known behaviour.

Ryanf55, 2024-11-23:
Can you elaborate where this is well documented? It's not in the warning section that plane will breach fences. 

https://ardupilot.org/plane/docs/common-polygon_fence.html

closed 2024-11-23 by Ryanf55

Hwurzburg, 2024-11-23:
> Can you elaborate where this is well documented? It's not in the warning section that plane will breach fences.
> 
> https://ardupilot.org/plane/docs/common-polygon_fence.html

suggestions for adding a warning there?

Ryanf55, 2024-11-23:
For me, I think the docs on fence could make a clear disclaimer that
* There are many situations to breach the fence, in most modes
* users should ensure that fences are sufficiently far from any obstacles because fences cannot currently be used for safety applications.
* Fence is completely disregarded in other modes, such as takeoff and RTL
* Test extensively in SITL before attempting to rely on fences to keep the vehicle safe

Here's another example of GUIDED flying straight until it hits the fence. AP could decide to turn before it hits the fence and never breach it.

[image]

It might be well known behavior for devs, but I don't think that's a good reason to not make this abundantly obvious to users that the fence implementation is not yet capable of keeping the vehicle inside the fence.

Hwurzburg, 2024-11-24:
the statement "Fence is completely disregarded in other modes, such as takeoff and RTL" is not true.....fences are active and work in those modes, fence actions are limited, however....RTLing as an action in RTL mode is pretty feeble, but it does work,just doesn't do much....RTL in takeoff works on breach...

" fences cannot currently be used for safety applications." is a misstatement..
fences can indeed be used for safety purposes if constructed correctly.....Copter AVOIDs fence breaches in many cases....Plane primarily takes action to mitigate fence breaches....the only Plane fences that attempt avoidance are altitude limits

so I think a bit more accurate wording could be added, and I will attempt it when I get a chance

timtuxworth, 2024-11-24:
> It might be well known behavior for devs, but I don't think that's a good reason to not make this abundantly obvious to users that the fence implementation is not yet capable of keeping the vehicle where the user configured it to fly.

I totally agree. You know you can PR the wiki @Ryanf55 ?

Ryanf55, 2024-11-24:
Henry - I've closed the issue with plane breaching fence in Takeoff because I can't reproduce it. 
Per the title, this issue is specific to Plane, which does not have the same capabilities as Fence. 
I am well aware that Copter has more intelligent behaviors. Thanks for the PR -it's an improvement!

Tim - no need to be sarcastic. I'm just trying to help ArduPlane users understand the fence limitations. By filing the issue, I am hoping for constructive feedback from other developers and users.

LupusTheCanine, 2024-11-24:
> Can you elaborate where this is well documented? It's not in the warning section that plane will breach fences.
> 
> https://ardupilot.org/plane/docs/common-polygon_fence.html

https://ardupilot.org/plane/docs/geofencing.html#:~:text=When%20fencing%20is,take%20back%20control.

Here, apparently it got yeeted into the archive since I first read that 🤷