ArduPilot/ardupilot #28576: install-prereqs-ubuntu.sh broken on RPi 4 running Ubuntu 24.04

bobzwik, 2024-11-10:
## Bug report

**Issue details**

Running `Tools/environment_install/install-prereqs-ubuntu.sh -y` hangs on an RPi 4 running Ubuntu 24.04. It gets stuck on `Building wheel for wxpython (setup.py)` and maxes out all my cores to 100%. 

To reproduce:
* Flash SD card with Ubuntu 24.04 for RPi 4 using Raspberry Pi imager.
* Boot Pi, set location, language, etc.
```
sudo apt update && sudo apt upgrade -y
sudo reboot
sudo apt install git
git clone --recurse-submodules https://github.com/ArduPilot/ardupilot.git 
cd ardupilot 
Tools/environment_install/install-prereqs-ubuntu.sh -y
```
I can also reproduce this without doing `sudo apt upgrade -y`

I don't know if this is linked, but after `sudo apt upgrade -y`, if i try to `update` and `upgrade` again, I get a message : 
```
The following upgrades have been deferred dure to phasing:
     python3-distupgrade shotwell shotwell-common ubuntu-release-upgrader-core ubuntu-release-upgrader-gtk 
```

**Version**
N/A

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
N/A

**Logs**
N/A

peterbarker, 2024-11-24:
Sorry, you have to leave it for a *lot* longer than 40 minutes for that step to complete....

It *does* complete.

Try again (leave it overnight....) and re-open if it reproduces (and I'll attempt to work things out locally).

closed 2024-11-24 by peterbarker

bobzwik, 2024-11-24:
Any reason why it takes that long? The whole `install-prereqs-ubuntu.sh` takes 5-8 minutes on the same RPi 4 on Ubuntu 22.04.

timtuxworth, 2024-11-24:
Make sure you have enough memory. I usually set my swap size to 2x physical memory. It usually solves this on the Pi.