ArduPilot/ardupilot #28586: LOG_REPLAY creating 100+ cores

h4writer, 2024-11-11:
## Bug report

**Issue details**

I created a log with LOG_REPLAY=1 and LOG_DISARMED=2 in order to better understand an issue I'm seeing. I hoped to be able to play with the code of EKF3 a bit. 

I've run "build/sitl/tool/Replay xxx.bin" which did work.

Now when looking at the log (using APM Planner, sorry oldschool) I see (C:0-102):
[image]

When reading the documentation I should only have C:0,1,2 and C:100,101,102

**Version**
Bin has been created with ardupilot4.4.4
Tried replay on 4.4.4 and 4.5.7, both gave this issue.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quadcopter

**Hardware type**
Cube orange

**Logs**
https://www.dropbox.com/scl/fi/x12kgwk3c3nlegpcx5ox8/24-11-08_13-15-02_1.bin?rlkey=dfp0v671n6jny4yerh5kyncw1&st=gg2ewjti&dl=0

Georacer, 2024-11-12:
Hello!

Perhaps you uploaded the wrong log?
When I open this one I see only 3 cores: 
[image]

h4writer, 2024-11-12:
The log I uploaded, is the input to the replay.

This is the output after running it through Replay:
https://www.dropbox.com/scl/fi/aoavck3nfyoawcpslz9oy/00000001.BIN?rlkey=4zbpxtalspb4iuxd4ij3pxwob&st=6bj85xfv&dl=0

Georacer, 2024-11-13:
It's probably a bug with APM Planner.

I'm seeing the correct cores with PlotJuggler:
[image]

h4writer, 2024-11-13:
Thanks for the plug to PlotJuggler. In PlotJuggler I also see the correct data!

closed 2024-11-13 by h4writer