ArduPilot/ardupilot #28589: inaccurate timestamps after a few days?

rmackay9, 2024-11-11:
[There's a report here that says we have a problem with timestamps](https://rsaxvc.net/blog/2024/2/4/ArduPilot_A_Matter_of_Time.html) becoming inaccurate over time.

This was brought up in [this discussion](https://discuss.ardupilot.org/t/sailboats-over-the-world/125973/6)

And PR https://github.com/ArduPilot/ardupilot/pull/26104 apparently resolved the issue in Linux

FYI @Georacer

rmackay9, 2024-11-12:
We had a good discussion about this on the dev call and we're looking forward to seeing logs

We think that AP should be able to run for many days and we have reports from at least two developers who are successfully running autopilots for months which suggests that the issue could be hardware specific. In particular we know that some autopilots suffer from severe clock drift.

Still, it's an important issue because we are looking forward to AP being used on boats crossing oceans

geofrancis, 2024-11-12:
is there any documentation on the clock drift issue with ardupilot?

geofrancis, 2024-11-12:
Based on the report that there are some controllers that have been on a desk running for extended periods of time the only things I can think of that would be different would be temperature and how much data is consuming if those controllers don't have a GPS connected.

I have an omnibusf4 set to offline logging at 1hz with a 32gb sd card in it. its plugged into a phone charger and il just leave it for a week. I have a clone pixhawk and a matekh743 I will also test with but i need to get more memory cards .

If that doesnt do anything then I will do the same test but with a GPS attached in case it has some effect on it. 

To test for clock drift issues then there I have ordered a TEC cooler to build a thermal cycling setup to take the controller from 0c to 60c repeatedly , hopefully that can highlight some issues if there is temperature related clock or gyro drift.

geofrancis, 2024-11-12:
I now have the omnibusf4 and the pixhawk clone flashed with latest stable rover and the h743 is on my 9 inch quad connected as im currently working on my other rovers.  I have set offline logging enabled at 1hz with a 32gb sd card on all of them and they are placed next to a window to get some variable temperatures so barring any power cuts I should have logs in a week,.

rmackay9, 2024-11-13:
@geofrancis,

I posted over in [this discussion](https://discuss.ardupilot.org/t/sailboats-over-the-world/125973/10) but for anyone else reading this issue here are my comments again

It’s too bad that the LOCARB person had such a bad experience. There’s no log to analyse and so it’s just a guess but it seems like his vehicle had estimation issues (e.g. EKF and/or DCM) and he struggled with AP’s complexity.

I’m keen to help anyone who wants to create a boat to cross a sea or travel around the world and I think AP can do it.

rmackay9, 2024-12-14:
@geofrancis,

I think it's OK for us to close this issue now right?  According to [the discussion](https://discuss.ardupilot.org/t/sailboats-over-the-world/125973/32) I think you tested for a few days and saw no issues

I'll go ahead and close this but I'm happy to re-open if we can reproduce it

closed 2024-12-14 by rmackay9