ArduPilot/ardupilot #28598: Error: The logical indices contain a true value outside of the array bounds.

Husseinhajjsleiman, 2024-11-12:
https://github.com/ArduPilot/ardupilot/blob/3bc0207f22d48648d0c46d301904fa782861733d/Tools/simulink/arducopter/functions/resampleLog.m#L76

Hi, 

I am using this [bin file](https://drive.google.com/drive/folders/1jDQYYb2LafGA-mumTwyLElMaQOSsPn41) for testing the simulink model. I am aware that this model is built for system identification mode however, I noticed that the code is capable of also handling different modes by simply selecting the subflight that I am interested in. 

Once the code start to resample the data (specifically when it reaches to BAT struct in logResampled) an error occurs showing the following: 
The logical indices contain a true value outside of the array bounds.
Error in resampleLog (line 76)
    timePlaus = plausibilizeTime(log.(prop).TimeS(idxVec));
Error in sid_pre (line 429)
    sidLogs(numOldSubflights + i).data = resampleLog(log, true, 'linear');

Is this error caused by the log bitmask? I have it set to 2242559

Thank you.

IamPete1, 2024-11-12:
@fbredeme can you take a look please.

Husseinhajjsleiman, 2024-11-14:
Anyone?

fbredeme, 2024-11-14:
Hi @Husseinhajjsleiman, what copter version are you using and what is the sample time of the BAT data?

Husseinhajjsleiman, 2024-11-15:
> Hi @Husseinhajjsleiman, what copter version are you using and what is the sample time of the BAT data?

Thank you for reaching back. I am using this ArduCopter V4.5.6 (7ce11b41) and its the bdshot version. 
I am not sure what sampling rate battery is in. This is a question that ive been looking for an answer for it on how to impact (increase or decrease) the sampling rate the dataflash logs (without synchronization or resampling)

I think its around 25 Hz but please double check after me. You can find the bin file in the link i shared in my first post. Note that I am using PM02D Holybro power module to measure battery params.

Husseinhajjsleiman, 2024-11-19:
any ideas why the error is occurring when Battery data getting resampled? I was debugging and I knew the error was occurring on battery logs. @fbredeme

fbredeme, 2024-11-19:
@Husseinhajjsleiman I need to set up Matlab again to check your bin file and how the code is handling it. I will reach back to you as soon as I managed to do so.
I looked at the logs in Mission Planner and it appears to me that the sample rate is 10 Hz. I will check that again in Matlab.
Please note that the last time that the Simulink model and the scripts have been updated was shortly before 4.3.

Husseinhajjsleiman, 2024-11-19:
Okay, thanks alot for reaching back. One question about sample rate: is there a way to increase the sample rate of the data flash logs? I am trying to impact the sample rate of position, velocities, gyros, imu and rpms - esc telem. Obviously they're from different sources and i am not sure if the attitudes are estimated. But i was wondering if you have an idea about increasing the sampling rate for those data sources? similar effect when sysid mode is activated and the data is sampled to 400 Hz. Any ideas are highly appreciated. I am looking forward to your comment about the error Im facing. Thanks.

Husseinhajjsleiman, 2024-11-25:
Hi, any updates about this issue @fbredeme ?

Zhujingxi, 2024-12-11:
Hi I'm also facing the smiler error can you share how you slove it?

Husseinhajjsleiman, 2024-12-12:
Honestly i thought that I fixed it but it worked with one case and I don't know why it worked with this one specific case. The person who implemented it must correct it as it is published in arducopter latest release. For some reason this is not clarified and no one is answering anymore. I am working on something else now but i will get back into it soon