ArduPilot/ardupilot #28612: ArduPilot 4.6 issues list

rmackay9, 2024-11-14:
This is a list of issues discovered during ArduPilot 4.6 testing.  Issues for [Copter/Rover-4.5 issues are here](https://github.com/ArduPilot/ardupilot/issues/26103)

**Reports requiring investigation**

- None

**Confirmed issues**

- None

**Enhancements or Issues that may not be resolved before the stable release**

- None

**Wiki issues/enhancement requests**

- None

amilcarlucas, 2024-11-15:
Why not copy the open issues from 4.5 into this page to make sure they are not forgotten?

rmackay9, 2024-11-15:
@amilcarlucas,

We could but I worry that the result will be that we have two really long lists of issues that we don't actually clear up.  If we can get focus from the dev team on investigating the issues and resolving them then that could work.  We should discuss at the next dev call I think

menschel, 2024-11-23:
@rmackay9 
In case you want to list it separately for plane.
Issue [Takeoff not completing under TKOFF_OPTIONS[0]=0 #28685 ](https://github.com/ArduPilot/ardupilot/issues/28685)
PR attempting to fix it [Plane: Fix inability to climb the last few meters of takeoff](https://github.com/ArduPilot/ardupilot/pull/28707)

rmackay9, 2024-11-25:
@menschel thanks for that, added!

menschel, 2024-11-29:
I have created another issue.

[Plane 4.6 beta - Takeoff throttle behavior for planes without airspeed sensor changed from 4.5.7](https://github.com/ArduPilot/ardupilot/issues/28775)

and the corresponding PR 
[Plane: Fix level-off and throttle without airspeed sensor](https://github.com/ArduPilot/ardupilot/pull/28776)

rmackay9, 2024-11-29:
Hi @menschel, thanks, added to the list!

MattKear, 2025-01-06:
There is a missing battery monitor option in the param meta data that will need to be brough across.  PR here:
#29017

timtuxworth, 2025-03-04:
This: Holybro M10 waiting to configure GPS1 ([discussion](https://discuss.ardupilot.org/t/fc-tries-to-config-gps-constantly-when-trying-to-arm-in-loiter-mode/129930/7), https://github.com/ArduPilot/ardupilot/pull/29320) -- resolved for -beta5

This fixed a problem I was seeing with the QioTek M10S GPS on master.

mikedkelly, 2025-03-10:
When I arm my plane the status changes to DCM Active and it throws continual Bad AHRS errors. As soon as I disarm it changes back to EKF3 Active.

Ryanf55, 2025-03-10:
> When I arm my plane the status changes to DCM Active and it throws continual Bad AHRS errors. As soon as I disarm it changes back to EKF3 Active.

Can you set LOG_DISARMED to 1 and get logs of it? Have you tried to reproduce on 4.5?

mikedkelly, 2025-03-10:
Here is the log and my params. I confirmed this happens on 4.5 and 4.6 beta 4. I guess I didn't notice it before.
[36 3-10-2025 10-31-44 AM.zip](https://github.com/user-attachments/files/19166847/36.3-10-2025.10-31-44.AM.zip)

rmackay9, 2025-03-11:
Hi @mikedkelly, @Ryanf55,

It's tough to do support on this issue so I've created a new discussion here https://discuss.ardupilot.org/t/mikedkellys-report-on-dcm-active-and-bad-ahrs/131592

timtuxworth, 2025-03-23:
There is definitely a problem with the M10S @tridge . I had a plane that was struggling to get satellites on it's M10S, so I replaced it with a brand new one out of the box. After 2 hours after booting with master no (zero satellites), I flashed 4.5.7 and it immediately came up with 11 satellites.

andyp1per, 2025-03-24:
> There is definitely a problem with the M10S [@tridge](https://github.com/tridge) . I had a plane that was struggling to get satellites on it's M10S, so I replaced it with a brand new one out of the [...]

I tend to agree - seeing issues on an M10 on my Pavo20.

timtuxworth, 2025-03-25:
> I tend to agree - seeing issues on an M10 on my Pavo20.

Try building the latest master Andy (more recent than 23rd March) - do you still see the problem?

rmackay9, 2025-03-26:
Hi @andyp1per @timtuxworth,

This issue probably isn't the best place to discuss this.  Can we move the discussion to [this new topic](https://discuss.ardupilot.org/t/timts-dronecan-ublox-m10-num-sats-issue/132353)?

ilia-enginer, 2025-07-17:
hello. I have a problem with using ardupilot sub of any version. after flashing usb firmware, the usb connection with the mission planner is lost. this is a bug in all stable versions of sub for speedebeef405v3. only the latest firmware version for developers using the speedebeef405v3 flight controller works properly