ArduPilot/ardupilot #28646: SR4__EXTRA1 won't work, SR2_EXTRA1 works instead

hoorkeet, 2024-11-16:
## Bug report

**Issue details**

Need a higher MavLink rate via UART, but changing SR4__EXTRA1 produce no effect, SR2_EXTRA1 works instead. Connected to Tx4/Rx4 physically, it maps to SERIAL4 and SERIAL4_PROTOCOL/SERIAL4_BAUD works ok, so I decided it is just a small bug?

**Version**
4.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[*] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Matek F405 WING V2

amilcarlucas, 2024-11-18:
SRx_parameters are meant for the x mavlink channel, not for the x serial port.

Also if serial 4 is your third mavlink channel SR2_* are the correct parameters. So not a bug.

rmackay9, 2024-11-18:
Looks like @amilcarlucas has resolved this so I'll close this if that's OK

closed 2024-11-18 by rmackay9