ArduPilot/ardupilot #28647: Copter:  Cannot ARM/DISARM in MODE2 using the game controller

muramura, 2024-11-16:
## Bug report

**Issue details**

Game controller PS4 DUALSHOCK 4 is calibrated in MODE 2 and does not ARM or DISARM when operating ARM or DISARM with the stick

**Version**

ArduCopter V4.6.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

QUAD/X

**Hardware type**

STAMPFLY

**Logs**

NONE

tpwrules, 2024-11-16:
The problem with game controllers is that they generally can't output minimum throttle and maximum yaw at the same time due to the circular aperture. I'm not sure this is Copter's fault, though the thresholds are fixed.

The software reading the controller and converting it into commands can calibrate or map differently to resolve this, though I don't know if any software supports that.

rmackay9, 2024-11-17:
OK, if @tpwrules is right then the controller is probably internally converting the stick's square movements into a circle.  AP also does this conversion and it's important because when you put the sticks in the corners you don't want the vehicle's lean angle to be ANGLE_MAX in both roll and pitch because that would result in a total lean angle of more than ANGLE_MAX

A solution could be to add an RC_OPTION to disable AP's conversion from square to circle

tpwrules, 2024-11-18:
The sticks can't move in a square, they can only move in a circle. I don't think they are converting anything. I also think this conversion happens after the check for arming. Can you point me to it?

rmackay9, 2024-11-18:
Hi @tpwrules,

It's in AP_Math's control.h/cpp in the rc_input_to_roll_pitch() function

peterbarker, 2024-11-23:
You should be able to arm in non-manual-throttle-modes if you mark the throttle as "sprung" using `PILOT_THR_BHV`

peterbarker, 2024-11-26:
Another thought - not only could you not arm because the yaw wasn't at the limit - but your throttle would also probably be outside the deadzone, and that's *not* something we should bypass!