ArduPilot/ardupilot #28662: AP_RangeFinder: add support for the LDRobot LD19

rmackay9, 2024-11-17:
AP already supports the LD-06 but a new model is available called the LD-19.  It's not clear if it uses the same protocol or not but if it's different we should update the driver to support the new model

This request comes from [this 4.6 beta testing discussion](https://discuss.ardupilot.org/t/copter-4-6-0-beta1-is-available-for-beta-testing/126295/9)

anand2532, 2025-01-04:
To contribute to LD19 rangefinder support, I propose:

1. Verification Steps
- Confirm protocol differences between LD-06 and LD-19
- Obtain official datasheet/communication specifications
- Compare existing LD-06 driver implementation

2. Implementation Strategy
- Examine current `AP_RangeFinder_LDRobot` driver in `/libraries/AP_RangeFinder/`
- Develop protocol compatibility check
- Create driver variant or extend existing implementation
- Implement device-specific configuration/initialization

3. Required Information
- Precise communication protocol details
- Supported baud rates
- Data frame structure
- Any unique initialization sequences

Requesting maintainer guidance on:
- Recommended approach for new model support
- Availability of official documentation
- Preferred testing methodology

Would appreciate insights to proceed effectively.

rmackay9, 2025-04-01:
@peterbarker,

I think we can close this now because PR https://github.com/ArduPilot/ardupilot/pull/29383 implies that we do have support for the LD19 now.

We still need the [LD06 wiki](https://ardupilot.org/copter/docs/common-ld06.html) to be updated though to at least mention that the LD19 also works.  Do you know of any parameter changes that are required compared to the LD06 to allow the LD19 to work?

closed 2025-04-22 by rmackay9