ArduPilot/ardupilot #28666: Fences ignored despite being enabled

Ryanf55, 2024-11-18:
## Bug report

**Issue details**

FENCE_ENABLE = 1
The plane is flying in TAKEOFF mode.
The plane is breaching the fence every rotation.
There is no indication in MAVProxy the fence is being breached.
MAVProxy shows `FEN` in gray, which I thought meant disabled.

**Version**
`master` today, hash f08a2120a8ce990e3939496dbe50ea1d4cae9fcb
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL

**Hardware type**
N/A
**Logs**

[logs.zip](https://github.com/user-attachments/files/17795178/logs.zip)
[fence-items.txt](https://github.com/user-attachments/files/17795183/fence-items.txt)

I found this trying to test the vehicle behavior with a non-[simple](https://en.wikipedia.org/wiki/Simple_polygon) polygon fence. 

**Screenshot**
[image]

Ryanf55, 2024-11-18:
I've reproduced this with a simple polygon (square) fence. 
[image]

From peter barker: `Not yet AFAIK.  It may even be two issues; in my case it might be a change in the edge-triggering of the RC fence switch.  In tridge's case it might be FENCE_AUTOENABLE not working.  In any case - not a <#715368769983348806> issue 🙂`

tridge, 2024-11-18:
@Ryanf55 in your screenshot fence is disabled:
[image]

Ryanf55, 2024-11-18:
> @Ryanf55 in your screenshot fence is disabled: [...]

Yes, but the parameter says it's enabled. It's conflicting info.

Ryanf55, 2024-11-24:
I can't reproduce this despite numerous attempts on the same hash with both inclusion and exclusion fences. 

I have found plenty of other fence issues, but not this specific one.

closed 2024-11-24 by Ryanf55