ArduPilot/ardupilot #28681: regression in plane takeoff behaviour in 4.6

tridge, 2024-11-19:
in 4.6.x we have introduced a bug in takeoff handling. We are not correctly handling an incorrect yaw estimate on takeoff. This is demonstrated in the following flight using CMAC_South, so the plane is pointing approximately south, with COMPASS_USE disabled:
[image]
here is the same test with 4.5.7:
[image]

menschel, 2024-11-20:
As reported following https://github.com/ArduPilot/ardupilot/pull/28352 there are three observed regressions in takeoff behaviour.

- The not excepted yaw reset by GPS leading to the course corrections mentioned above.
- The continuous use of 100% throttle after `TKOFF_THR_MAX_T` is exceeded until `TKOFF_ALT` or the mission items takeoff altitude is reached. Previously it fell back to TECS controlled throttle based on the pitch angle.
- The level-off functionality can level out the plane below target altitude thus preventing takeoff to complete.

menschel, 2024-11-29:
The YAW issue is not present in auto takeoff mission item. 
Measurements are referenced in the PR for the items I mentioned.
https://github.com/ArduPilot/ardupilot/pull/28776