ArduPilot/ardupilot #28685: Takeoff not completing under TKOFF_OPTIONS[0]=0

Georacer, 2024-11-20:
User @menschel reported that
- in a Plane
- without an airspeed sensor -> running under `TKOFF_OPTIONS[0]=0`
- in an AUTO mission with a takeoff item to 50m
the aircraft would climb to 47m and just hang there, not completing the mission item.
[image]
The final reach of 50m was accidental.

He was flying a small ZOHD flying wing.
He kindly provided a flight log for us [here](https://github.com/ArduPilot/ardupilot/pull/28352#issuecomment-2482853648).

He attempted to file a fix here: https://github.com/ArduPilot/ardupilot/pull/28352, but what he proposes isn't the solution.

This affects our 4.6.0 beta.

---

While it was unable to climb the last 3m, the airframe was at full throttle and at ~6deg pitch up (3deg nav pitch + 3deg `PTCH_TRIM_DEG`)
Still, that wasn't enough to make the last few meters.

tridge, 2024-11-20:
we discussed on the EU dev call, and think we should gradually bring in TECS control during the takeoff, with a widening pitch window available to TECS, so in initial part of takeoff the TECS min/max pitch would be completely constrained, then it widens during the takeoff so that at the end of the takeoff TECS has full control

Georacer, 2024-11-20:
> Just for clarification.
It's a plane with a tail [1](https://www.zohd.net/zohd-drift), [2](https://oscarliang.com/zohd-drift-dark-breeze/) not a flying wing, customized sub 250g build (246g)

Thanks!

Georacer, 2025-04-29:
This is now addressed by #28792.

closed 2025-04-29 by Georacer