ArduPilot/ardupilot #28712: Plane: Pullup passes wait_airspeed stage without ever meeting airspeed criteria

ohitstarik, 2024-11-23:
## Bug report

**Issue details**

The new bugfix of pullup (https://github.com/ArduPilot/ardupilot/commit/1c194878ee6d527ba30cc35c44d4780f79dbc1cc ) changes the behavior of stage wait_airspeed such that the stage passes if the desired airspeed is reached or pitch is above pitch_min. This is meant to cover a scenario where pitch trim is bad enough that the airframe pulls up before the desired airspeed is reached, most likely leading to stalls, or the wait_airspeed stage never passing. 

However, more concerningly, this introduces situations where if the aircraft is in a freefall and:
- gets caught in a line
- is released from anywhere but the tail
- tumbles due to wind (lower altitude testing)
- or is in any other situation where it meets the criteria for pitch_min, 

wait_airspeed is passed, and the vehicle begins pulling up before it is allowed to gain enough airspeed to do so, or have a stable enough attitude to do so, both of which are critical to a successful pullup maneuver at high altitudes.

I believe this bugfix needs to be reworked, or rolled back. I don't see a situation where a team would send a vehicle up to altitude with bad pitch trim, or without having tested the vehicle and its pitch trim numerous times on the ground. Meanwhile, an in-flight error is spontaneous and could happen at any time, allowing pitch_min to be reached and causing errors during the pullup maneuver. 

**Version**
4.6.0 beta

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

lzones-canada, 2024-11-30:
We just got bit by this bug in real flight past week. Glider did not release from Balloon initially and pitch sensor triggered the WAIT_AIRSPEED - I am glad you brought this up. We have reverted since.

DjMixMasterDragon, 2024-11-30:
Can confirm this issue - we just got bit by this bug in real flight past week. Glider did not release from balloon initially and pitch sensor triggered the WAIT_AIRSPEED - I am glad you brought this up. We have reverted since.