ArduPilot/ardupilot #28720: Changing SIM_IMU_POS_X/Y causes the Copter to deviate from its original flight path during turns and crash

wyunc, 2024-11-24:
## Bug report

**Issue details**

When I set up a rectangular flight mission and set SIM_IMU_POS_X to 50, I noticed that at the corners of the rectangle, the drone suddenly switched to Land mode, significantly deviated from the flight path, and ultimately crashed. (Reproduction: `param set SIM_IMU_POS_X 50`)

As shown in the figure below, while flying along the vertical yellow flight path to a waypoint and making a right-angle turn, the drone was expected to continue flying leftwards. However, it actually deviated towards the bottom-right of the map (creating an arc in the air).
[image]
[image]

In the end, it significantly deviated from the original flight path and crashed.
[image]

Similarly, we wondered if the same issue would occur with SIM_IMU_POS_Y. We found that when SIM_IMU_POS_Y was set to 50, the drone also deviated towards the bottom-left of the map during right-angle turns.
[image]
[image]

I am unsure whether this is a bug or an issue with my settings. I am more inclined to believe this is a bug because there is no [documentation](https://ardupilot.org/copter/docs/parameters.html#sim-imu-pos-x-imu-offsets) imposing restrictions on these parameters.

**Version**
4.6.0

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
SITL

**Logs**
[hitground.zip](https://github.com/user-attachments/files/17892762/hitground.zip)

wyunc, 2024-11-24:
Additionally, when SIM_IMU_POS_Z is set to 50, the drone suddenly disconnects just before reaching a turning point, and its position on the map stops updating. This behavior is entirely different from the effects of X/Y, so I believe this might be another type of bug?
[image]

IamPete1, 2024-11-24:
I don't think its reasonable to expect AP to work with a 50 meter IMU offset. Try 5 meters or less.

wyunc, 2024-11-24:
Hi @IamPete1,
I set SIM_IMU_POS_X = 5 and found that the issue still exists. The degree of deviation becomes smaller, but the drone still ends up hit the ground. However, when SIM_IMU_POS_X is set to 1, the issue no longer occurs, although there is still some deviation from the flight path.
[image]
[image]

IamPete1, 2024-11-24:
Did you also set INS_POS1_X, INS_POS2_X, INS_POS3_X? See:
https://ardupilot.org/plane/docs/common-sensor-offset-compensation.html

wyunc, 2024-11-24:
> Did you also set INS_POS1_X, INS_POS2_X, INS_POS3_X? See: https://ardupilot.org/plane/docs/common-sensor-offset-compensation.html

No, their values are all 0. The only parameter I changed was SIM_IMU_POS_X/Y/Z; all other parameters remain at their default values. You can find the log data for both(SIM_IMU_POS_X=50 and SIM_IMU_POS_Y=50) scenarios in the hitground.zip file I provided.

IamPete1, 2024-11-24:
If you have a 5 meter position offset and you don't tell AP about it I think that is a miss configuration. Hopefully that will fix it.

rmackay9, 2024-11-24:
Thanks for the report.

I'm going to go ahead and close this because it's really a developer support issue and we just need to try and keep our to-do list (which is what the issues list is) free of support calls.  We will still see replies though.  Txs again for using AP!

closed 2024-11-24 by rmackay9

rmackay9, 2024-11-29:
Hi @wyunc,

I think it's best to do testing in "master" which is now 4.7.0-dev because this is the active branch and is where any bugs would be fixed first.  Normally any fixes are applied to "master" and then back-ported to the latest beta or stable branch.

wyunc, 2024-11-29:
> Hi @wyunc,
> 
> I think it's best to do testing in "master" which is now 4.7.0-dev because this is the active branch and is where any bugs would be fixed first. Normally any fixes are applied to "master" and then back-ported to the latest beta or stable branch.

Got it, thanks a lot!