ArduPilot/ardupilot #28757: VTX power level on Foxeer Reaper fails and pit mode isn't working

IvanLeu, 2024-11-27:
Cannot set 2500mw on Foxeer Reaper Extreme with IRC Tramp protocol. I am using MatekH743 with ArduPlane v4.5.6. I noticed that pitmode isn't working and power levels 25mw and 400mw work fine, but 2500mw isn't working. Btw I added 2500 power level in vtx power table.  By the way It seems to work on Betaflight firmware.

On IFlight Blitz 5.8G it works perfectly and pit mode works.

I turned on debug messages and got following output. Seems like tramp sends command to set 2500 and gets the response that active power is still 400 or 25.
[image]

Also on IFlight Bltiz 1.2G pit mode still fails which I feel is very strange because 5.8G works.

How can I solve this problem? The fact that it works perfectly on 5.8G tells me that is the problem is not in the firmware code, but i cant really tell.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2024-11-27:
I think this is because the VTX lies about the power levels it supports. Try setting VTX_MAX_POWER to 600

IvanLeu, 2024-11-27:
> I think this is because the VTX lies about the power levels it supports. Try setting VTX_MAX_POWER to 600

It tried different max powers, 1000, 1600 and 2500 - same result. I am sure it should support one of them.

IvanLeu, 2024-11-29:
[image]
what could cause cush behaviour of smart audio?

IvanLeu, 2024-11-30:
> I think this is because the VTX lies about the power levels it supports. Try setting VTX_MAX_POWER to 600

That worked

closed 2024-11-30 by IvanLeu