ArduPilot/ardupilot #28759: H743 flash corruption with brownout.

geofrancis, 2024-11-27:
From my testing the H743 chip can corrupt its flash memory if its power supply doesnt come up fast enough. I can make matek boards corrupt the memory by changing the voltage between 0 and 5v slowly. my understanding is that the cube carriers have an extra power management chip so it doesn't start up until the voltage is adequate.

When this happens the board appears dead since reflashing in DFU mode doesnt fix it, you need to do a full flash erase with STM32CUBE programmer and reflash the board to get it to start up again.

[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

geofrancis, 2025-04-17:
Faulty early matek boards.

rmackay9, 2025-04-17:
Thanks @geofrancis for the update!

hendjoshsr71, 2025-06-05:
This seems completely reasonable given my recent testing and the 4.6 changes that wipe firmware on any error.

Also likely reasonably repeatable across all H7 .

Note I’ve also seen I may have possibly seen this in the same exact hardware using Betaflight. Making me lean towards a hardware power supply cause.

geofrancis, 2025-06-05:
how is it done on linux because i think i might have the same issue there on my beaglebone blue

geofrancis, 2025-06-05:
how is it checked on linux? as i have some similar issues on my beaglebone where i had to wipe it but it's not identical so im not sure if its related.

hendjoshsr71, 2025-06-05:
@geofrancis what do you mean by DFU? What be sure we have the same definition or process?

geofrancis, 2025-06-05:
hold the boot button in and power it on. then flash ardurover with bl

geofrancis, 2025-06-05:
im not an expert but I think its something to do with it doesnt overwrite the data if it thinks there is nothing there, so the old data remains, I had something similar on esp32, im not sure if stm32 works the same.

geofrancis, 2025-06-05:
I think the linux issue is hardware related to the flash on the beaglebone corrupting and not connected to this issue other than just similar results.