ArduPilot/ardupilot #28761: hwdef inconsistencies

amilcarlucas, 2024-11-27:
## Bug report

**Issue details**

The APJ_BOARD_ID, USB_STRING_MANUFACTURER and USB_STRING_PRODUCT have all sorts of inconsistencies:

VID 0x1209 has different vendor names: ArduPilot and AET
VID 0x1209 has different vendor names: ArduPilot and Airvolute
VID 0x1209 has different vendor names: ArduPilot and Aocoda-RC-H743Dual
VID 0x1209 has different vendor names: ArduPilot and CSKY
Warning: Duplicate APJ board ID 50 for different boards: CUAVv5Nano and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: CUAVv5Nano and CUAVv5
Warning: Duplicate APJ board ID 9 for different boards: CubeBlack and CUAV-Pixhack-v3
AP_IOMCU_PROFILED_SUPPORT_ENABLED already in defines with same value
VID 0x2DAE PID 0x1011 has different product names: CubeBlack and CubeGreen-solo
Warning: Duplicate APJ board ID 9 for different boards: CubeGreen-solo and CUAV-Pixhack-v3
HAL_MONITOR_THREAD_ENABLED already in defines with same value
Warning: Duplicate APJ board ID 1079 for different boards: CubeNode-ETH and CubeNode
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE PID 0x1016 has different product names: CubeOrange and CubeOrange-periph
Warning: Duplicate APJ board ID 1400 for different boards: CubeOrange-periph and CubeOrange
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
Warning: Duplicate APJ board ID 9 for different boards: CubePurple and CUAV-Pixhack-v3
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
AP_NETWORKING_BACKEND_PPP already in defines with same value
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
Warning: Duplicate APJ board ID 9 for different boards: CubeSolo and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 1000 for different boards: f103-Airspeed and f103
Warning: Duplicate APJ board ID 1000 for different boards: f103-HWESC and f103
Warning: Duplicate APJ board ID 1000 for different boards: f103-QiotekPeriph and f103
Warning: Duplicate APJ board ID 1000 for different boards: f103-RangeFinder and f103
Warning: Duplicate APJ board ID 1000 for different boards: f103-Trigger and f103
Warning: Duplicate APJ board ID 1004 for different boards: f303-HWESC and f303
Warning: Duplicate APJ board ID 1004 for different boards: f303-M10025 and f303
Warning: Duplicate APJ board ID 1004 for different boards: f303-M10070 and f303
Warning: Duplicate APJ board ID 1004 for different boards: f303-MatekGPS and f303
Warning: Duplicate APJ board ID 1004 for different boards: f303-PWM and f303
Warning: Duplicate APJ board ID 1004 for different boards: f303-TempSensor and f303
Warning: Duplicate APJ board ID 1004 for different boards: f303-Universal and f303
Warning: Duplicate APJ board ID 1014 for different boards: f405-MatekGPS and f405-MatekAirspeed
VID 0x27AC PID 0x0201 has different product names: F4BY and F4BY_F427
Warning: Duplicate APJ board ID 9 for different boards: fmuv2 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: fmuv3 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: fmuv3 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 50 for different boards: fmuv5 and CUAVv5
Warning: Duplicate APJ board ID 1027 for different boards: G4-ESC and FlywooF745
Warning: Duplicate APJ board ID 146 for different boards: H757I_EVAL_intf and H757I_EVAL
Warning: Duplicate APJ board ID 1119 for different boards: HEEWING-F405v2 and HEEWING-F405
Warning: Duplicate APJ board ID 1043 for different boards: Here4FC and Here4AP
Warning: Duplicate APJ board ID 1053 for different boards: HolybroG4_GPS and CarbonixL496
Warning: Duplicate APJ board ID 3 for different boards: iomcu-dshot and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-f103 and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-f103-8MHz-dshot and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-f103-dshot and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu_f103_8MHz and iomcu
VID 0x1209 has different vendor names: ArduPilot and Japan Aviation Electronics Industry Ltd.
Warning: Duplicate APJ board ID 1059 for different boards: JHEMCU-GSF405A-RX2 and JHEMCU-GSF405A
Warning: Duplicate APJ board ID 1058 for different boards: KakuteH7Mini-Nand and KakuteH7Mini
Warning: Duplicate APJ board ID 1048 for different boards: KakuteH7v2 and KakuteH7
Warning: Duplicate APJ board ID 1038 for different boards: MambaF405US-I2C and MambaF405-2022
Warning: Duplicate APJ board ID 1014 for different boards: MatekF405-CAN and f405-MatekAirspeed
Warning: Duplicate APJ board ID 125 for different boards: MatekF405-STD and MatekF405
Warning: Duplicate APJ board ID 143 for different boards: MatekF765-Wing and MatekF765-SE
Warning: Duplicate APJ board ID 143 for different boards: MatekF765-Wing and MatekF765-SE
Warning: Duplicate APJ board ID 1170 for different boards: MatekG474-Periph and MatekG474-DShot
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-Airspeed and MatekL431
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-APDTelem and MatekL431
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-BattMon and MatekL431
[... 47 lines cut ...]
Warning: Duplicate APJ board ID 9 for different boards: skyviper-journey and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: skyviper-v2450 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 50 for different boards: Swan-K1 and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: TBS-Colibri-F7 and CUAVv5
VID 0x27AC has different vendor names: Swift-Flyer and Laser Navigation

**Version**
master

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Not applicable

**Hardware type**
All supported boards

**Logs**
Posted above

tpwrules, 2024-11-28:
FWIW most/all of those MatekL431 should have the same board ID. They are all basically hardware compatible, you just flash the right one depending on which device it is and which features you want.

peterbarker, 2025-04-29:
@amilcarlucas please filter your list to the ones which are actual problems.

Note that we are very aware of some of the problems here - see https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/validate_board_list.py#L53 .  There's not a lot we can do about those - some of them could be fixed upstream if you can convince them to take a patch.

amilcarlucas, 2025-04-29:
The validate_board_list list looks incomplete to me.

Here is the filtered list:

AP_IOMCU_PROFILED_SUPPORT_ENABLED already in defines with same value
AP_NETWORKING_BACKEND_PPP already in defines with same value
HAL_MONITOR_THREAD_ENABLED already in defines with same value
HAL_MONITOR_THREAD_ENABLED already in defines with same value
VID 0x1209 has different vendor names: ArduPilot and AET
VID 0x1209 has different vendor names: ArduPilot and Airvolute
VID 0x1209 has different vendor names: ArduPilot and Aocoda-RC-H743Dual
VID 0x1209 has different vendor names: ArduPilot and CSKY
VID 0x1209 has different vendor names: ArduPilot and Dheeran labs
VID 0x1209 has different vendor names: ArduPilot and Dheeran labs
VID 0x1209 has different vendor names: ArduPilot and Horizon31
VID 0x1209 has different vendor names: ArduPilot and Japan Aviation Electronics Industry Ltd.
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and mRo
VID 0x1209 has different vendor names: ArduPilot and Qiotek
VID 0x1209 has different vendor names: ArduPilot and Sierra Aerospace
VID 0x1209 has different vendor names: ArduPilot and SkySakura
VID 0x27AC has different vendor names: Swift-Flyer and Laser Navigation
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
VID 0x2DAE has different vendor names: Hex/ProfiCNC and CubePilot
Warning: Duplicate APJ board ID 3 for different boards: iomcu_f103_8MHz and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-dshot and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-f103 and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-f103-8MHz-dshot and iomcu
Warning: Duplicate APJ board ID 3 for different boards: iomcu-f103-dshot and iomcu
Warning: Duplicate APJ board ID 3 for different boards: mini-pix and iomcu
Warning: Duplicate APJ board ID 9 for different boards: CubeBlack and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: CubeGreen-solo and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: CubePurple and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: CubeSolo and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: fmuv2 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: fmuv3 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: fmuv3 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: mRoX21 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: Pixhawk1 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: Pixhawk1 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: Pixhawk1-1M and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: Pixhawk1-1M and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: skyviper-f412-rev1 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: skyviper-journey and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 9 for different boards: skyviper-v2450 and CUAV-Pixhack-v3
Warning: Duplicate APJ board ID 11 for different boards: Pixracer and luminousbee4
Warning: Duplicate APJ board ID 11 for different boards: Pixracer and luminousbee4
Warning: Duplicate APJ board ID 50 for different boards: CUAVv5Nano and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: CUAVv5Nano and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: fmuv5 and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: PH4-mini and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: PH4-mini and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: Pix32v5 and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: Pixhawk4 and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: Pixhawk4 and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: Swan-K1 and CUAVv5
Warning: Duplicate APJ board ID 50 for different boards: TBS-Colibri-F7 and CUAVv5
Warning: Duplicate APJ board ID 124 for different boards: revo-mini-i2c and revo-mini
[... 30 lines cut ...]
Warning: Duplicate APJ board ID 1059 for different boards: JHEMCU-GSF405A-RX2 and JHEMCU-GSF405A
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-Airspeed and MatekL431
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-APDTelem and MatekL431
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-BattMon and MatekL431
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-DShot and MatekL431
Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-EFI and MatekL431
[... cut ...]
Warning: Duplicate APJ board ID 1170 for different boards: MatekG474-Periph and MatekG474-DShot
Warning: Duplicate APJ board ID 1400 for different boards: CubeOrange-periph and CubeOrange

There are more duplicated here than just the ones in the board_id_whitelist variable.

These warnings look strange to me:
AP_IOMCU_PROFILED_SUPPORT_ENABLED already in defines with same value
AP_NETWORKING_BACKEND_PPP already in defines with same value
HAL_MONITOR_THREAD_ENABLED already in defines with same value
HAL_MONITOR_THREAD_ENABLED already in defines with same value

peterbarker, 2025-05-09:
> Warning: Duplicate APJ board ID 1062 for different boards: MatekL431-Periph and MatekL431

As @tpwrules mentions - this isn't a problem with those sharing the same board ID.  You can flash *all sorts* of different firmwares to that board :-)

To some extent this is a problem where our hwdefs aren't just defining hardware but features to build into the firmware we run on the hardware.  Our entire include system is a bit of a kludge to get around the lack of separation around the concepts.
Pro-tip: hardware doesn't change between hwdef.dat and hwdef-bl.dat :-)

peterbarker, 2025-05-09:
> These warnings look strange to me:
> AP_IOMCU_PROFILED_SUPPORT_ENABLED already in defines with same value
> AP_NETWORKING_BACKEND_PPP already in defines with same value
> HAL_MONITOR_THREAD_ENABLED already in defines with same value
> HAL_MONITOR_THREAD_ENABLED already in defines with same value

Yeah, that's a bit of linting that could be done; it means that a board which is including X is setting a define to the same value that the thing it is including has already set it to - so you can remove the define in the board doing the including.

amilcarlucas, 2025-05-12:
The VID code is the same and that is confusing. What we should strive to do is provide a way to ask the board what is the vendor string. And would be nice to have a product string as well.

Then and GCS would be able to unequivocally display the hardware vendor. And product string would be great too.
We misused the USB PID and VID. They are massively duplicated.